e2nIEE/pandapower #1351: Timeseries - OutputWriter fails to init from log variables for '3ph" cases

JimiC, 2021-09-30:
When trying to resolve a `3ph` timeseries for a `net` while passing the variables we want to monitor via the `log_variables` i.e. `OutputWriter(net, log_variables=_variables)` were `_variables`  is a `dict` containing the `key, values` data of 
```
{
    "res_bus_3ph": ["vm_a_pu", "va_a_degree",
                    "vm_b_pu", "va_b_degree",
                    "vm_c_pu", "va_c_degree",
                    "p_a_mw", "q_a_mvar",
                    "p_b_mw", "q_b_mvar",
                    "p_c_mw", "q_c_mvar",
                    "unbalance_percent"],
    "res_ext_grid_3ph": ["p_a_mw", "q_a_mvar",
                         "p_b_mw", "q_b_mvar",
                         "p_c_mw", "q_c_mvar"],
...                         
```

the `OutputWriter` fails to initialize the `log_variables` when it tries to find "how many elements there are in net".
The problem lies specifically in https://github.com/e2nIEE/pandapower/blob/develop/pandapower/timeseries/output_writer.py#L407
where the `table.split("res_")[-1]` doesn't take into consideration the `_3ph` suffix we have in `3ph` cases.

**Solution:** Adjust `table.split("res_")[-1]` to `table.split("res_")[-1].split("_3ph")[0]` and problem solved. It works for both cases (balanced and asymmetric)

Let me know if you want me to provide a PR to resolve this issue.

Thanks in advance.

JimiC, 2022-02-27:
@eddie-atkinson just seen that the code has been modified to `table.split("res_")[-1].replace("_3ph", "")`. I guess this works as well. Will try it out asap and will report back.

LiesDebr, 2022-02-28:
Hey, I don't know if I am allowed to ask this here, but I try to run a timeseries with asymmetric loads en sgens. But this won't work. I had a question posted yesterday about this where i also posted my code. Is this how you run this too? My question: #Is it possible to use timeSeries calculation with unbalanced loads and sgens? in addition also using a storage? #1515

JimiC, 2022-03-02:
@eddie-atkinson I tested the fix and it's working. For reference it was fixed in 99f64f894b8b31ff1fd3c3707c8ccc936519c30c.
Thanks a lot.

/cc @gourab2009

JimiC, 2022-03-02:
@LiesDebr As long as you use `create_asymmetric_sgen` and `create_asymmetric_load` it should work.
See [pp docs](https://pandapower.readthedocs.io/en/develop/elements.html) for more info.

closed 2022-03-02 by JimiC

eddie-atkinson, 2022-03-02:
No worries @JimiC, happy to help