e2nIEE/pandapower #1384: time series for multiple elements represented with one controller

victoria274, 2021-11-09:
Hello,
I have a problem running a time series calculation for multiple loads controlled by one controller. Is there a way to fix this so it can run?
I have 27 loads total and the data frame draws data from a csv file "AUG_consumer_data_csv2" (attached), the time series portion is like this:

df=pd.read_csv("AUG_consumer_data_csv2.csv", index_col=0)
profiles = pd.DataFrame()
profiles["Weekdays kW"] = df["Weekdays kW"].tolist() 
print(df["Weekdays kW"].tolist())
ds = DFData(profiles)
for i in range(27):
    ConstControl(net, element='load', variable='p_mw', element_index=[i],
               data_source=ds, profile_name=["Weekdays kW"])
ow = OutputWriter(net, time_steps=48, output_path ="./ts_output", output_file_type = ".csv")
ow.log_variable('res_load', 'p_mw')
pp.timeseries.run_time_series.run_timeseries(net, time_steps=48)

[AUG_consumer_data_csv2.csv](https://github.com/e2nIEE/pandapower/files/7506331/AUG_consumer_data_csv2.csv)

thank you

rbolgaryn, 2021-11-10:
Dear @victoria274 ,

have you looked at the timeseries tutorial?

https://github.com/e2nIEE/pandapower/blob/develop/tutorials/time_series.ipynb

Roman

rbolgaryn, 2021-11-10:
For using a single ConstControl for multiple elements at once, pass a list or array for the following elements:

profile_name
element_index

victoria274, 2021-11-18:
I have an array pass the initial p value to all 27 loads like this
loadpmw = np.ones(27)*0.001453

I also pass the column of my csv data source file to a list like this
profiles["Weekdays kW"] = df["Weekdays kW"].tolist() 

I changed the ConstControl like you suggested with these values as arrays
ConstControl(net, element='load', variable='p_mw', element_index=loadpmw, data_source=ds, profile_name=["Weekdays kW"])
Without the for loop this time, a single controller for multiple elements. But this causes an error and the time series does not complete. 
Am I passing the elements wrong?

Also if I run the initial code I sent a week ago, the time series runs smoothly and I get a result from the output writer in a csv file. All the results to the output writer file are the same though, is that how it was supposed to be the result due to the fact that I give the same initial value to the loads? I am attaching the file below.
[p_mw.csv](https://github.com/e2nIEE/pandapower/files/7565739/p_mw.csv)
.

victoria274, 2021-11-18:
Another thing i have a question about is this note that I read in the documentation
"If multiple elements are represented with one controller, the data source must have integer columns. At the moment, only the DFData format is tested for the multiple const control." 

This means that all the data from the data source used must be integers, it will have an error if it has float numbers? Because from my data source csv file I have strings as the column's names and float as the data that is being read into the controller

rbolgaryn, 2021-12-03:
closing this issue because another issue is open #1404

closed 2021-12-03 by rbolgaryn