e2nIEE/pandapower #1387: AC power flow converge but ACOPF not converge

xfangnrel, 2021-11-11:
Hi,
I have a case which can have a converged AC power flow but the AC OPF can't converge. Do you know what might be the reason?

--> 357     _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
    358 
    359 

c:\users\xfang\gitrepos\pandapower-master\pandapower\optimal_powerflow.py in _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
     68 
     69     if not result["success"]:
---> 70         raise OPFNotConverged("Optimal Power Flow did not converge!")
     71 
     72     # ppci doesn't contain out of service elements, but ppc does -> copy results accordingly

OPFNotConverged: Optimal Power Flow did not converge!

Thanks a lot,
Xin

xfangnrel, 2021-12-28:
This issue has been solved. Thanks.

closed 2022-01-09 by rbolgaryn

fgo97, 2022-01-13:
How did you solve this? 
Thanks a lot, Franz