e2nIEE/pandapower #1389: State Estimation - Low Nodal Voltages

bbadetcauk, 2021-11-12:
Hello,

I have a pandapower model of a electrical network and I'm performing state estimation with convergence and relatively good accuracy.

The issue I'm having is that the volts across the network are consistently lower than the voltage magnitude measurement inputs (let's say 0.97pu rather than 1.01pu).

I have tried artificially adjusting the tap changers at the bulk supply point transformers, but any change in voltage is reflected on the bus on the HV side of the trafo (where there is a network infeed element), rather than on the LV side (connected to the primary network with majority of voltage measurements). 

How can I make the HV system 'stronger' in state estimation models, such that changes in tappings at the bulk supply point transformers regulate the primary voltage across the network?

 Many thanks in advance.

wangzhenassd, 2021-11-22:
Hi @bbadetcauk, I would suggest that, if the voltage are constantly lower than the expected value, you can try to adjust the std_dev of the voltage measurement. With a smaller std_dev, the v measurements carries higher weight in the state estimation process. After this step is done, you can then try to find the correct tap position so that the HV side of the trafo gets also the correct v.

vogt31337, 2025-04-07:
closed due to inactivity. Please make a new issue if the problem still persists and please provide a small example showing your problem.

closed 2025-04-07 by vogt31337