e2nIEE/pandapower #1396: Improve measurements of execution time

BDonnot, 2021-11-17:
Hello,

By digging a bit in the pandapower source code, i found out that the execution time (among others I image) are measured using `time.time()` which is not the most suited one. The documentation (taken from the documentation https://docs.python.org/3/library/time.html#time.time) states that:

> Note that even though the time is always returned as a floating point number, not all systems provide time with a better precision than 1 second. While this function normally returns non-decreasing values, it can return a lower value than a previous call if the system clock has been set back between the two calls.

And of course having a negative execution time might not be great.

The "fix" for this "issue" is to use the `perf_counter` of the same module (https://docs.python.org/3/library/time.html#time.perf_counter) which is used exactly in the same way.

Note however that is it not relevant to use `perf_counter` for measuring date and times, but only for measurement of elapsed time (as the name suggests)

Note: if you really want to measure really small duration, you can even use https://docs.python.org/3/library/time.html#time.perf_counter_ns, but then you'll have to manually divide the results by 1e8 because the elapsed time is given in nano seconds for this specific clock.

The" quick and dirty" way to "fix" this would be to change the import like `from time import time` to `from time import perf_counter as time` that won't change the code too much. However, personnally, I would advise to change the code however, and explicitely write `t0 = perf_counter ()` and `et = perf_counter () - t0` (example in the `_run_newton_raphson_pf` function)

If this changes suits you, i can make it and make a pull request if you want (just tell me on which branch to start and on which to do the merge request)

Benjamin

SteffenMeinecke, 2021-11-18:
yes, suits (I prefere also the more explicitely).

BDonnot, 2021-11-18:
I'm not sure if you plan on doing it or not.
I have no problem to do it, just let me know from which branch to start and i'll do the PR asap :-)

SteffenMeinecke, 2021-11-19:
yes, @BDonnot please raise a PR (from develop branch). That would be great!

BDonnot, 2021-11-19:
OK I'll do it as soon as I can.

closed 2021-11-23 by SteffenMeinecke