e2nIEE/pandapower #1397: Allow the possibility to use (or not) lighsim2grid even when it's installed

BDonnot, 2021-11-17:
Hello,

In the `run_newton_raphson_pf.py` file there is:

```python
try:
    from lightsim2grid.newtonpf import newtonpf as newton_ls
    lightsim2grid_available = True
except ImportError:
    lightsim2grid_available = False
```

Which is really nice and allow to use a faster c++ code  (lightsim2grid). However, it would be nice to have the possibility to use it or not, for example specified in the `options` that would evaluate to `True` by default.

In my opinion, this would act as the `numba` option when `runpf` is called: by default pandapower attempts to use numba, but there is an option to deactivate it if we don't want it.

In the code of `run_newton_raphson_pf.py` this could change to something like:

```python
    # run the newton power flow
    if lightsim2grid_available and options["use_lightsim2grid"]:
        V, success, iterations, J, Vm_it, Va_it = newton_ls(Ybus, Sbus, V0, pv, pq, ppci, options)
    else:
        V, success, iterations, J, Vm_it, Va_it = newtonpf(Ybus, Sbus, V0, ref, pv, pq, ppci, options)
```

A more refined change could also raise a warning if lightsim2grid is not installed (just like it does for numba) but i'm not sure that's a good idea (FYI lightsim2grid is now on pypi).

Benjamin

ascheidl, 2021-11-18:
I think its a good idea.

That lightsim2grid is now on pipy and easily installed on windows is really nice! Does this mean that it will replace pandapower as the default backend of grid2op ?

I saw that L2G offers some speedups for certain types of investigations. For example, many loadflows while the load is assumed to be constant (contigency analysis) or topology is constant (time series).  It would be nice if we could make these available in pp, too. However, in pp we have two additional steps beside the loadflow calculation:  to translate the pandas tables to the ppc structure and the results back in the tables. This translation makes an substancial part of the whole runpp (depending on grid size). Currently, the implementation of this part is complex and inefficient. It is probabily not easy to, for example, only translate new values for loads. However, I think the whole part can be implemented a lot faster using numba or c/c++. A new implementation could also allow for the necessary partly translation.

BDonnot, 2021-11-18:
Hello,

> That lightsim2grid is now on pipy and easily installed on windows is really nice! Does this mean that it will replace pandapower as the default backend of grid2op ?

No it will not. Pandapower will remain the default as it has more capabilities at the moment (even though it's slower)

> I saw that L2G offers some speedups for certain types of investigations. For example, many loadflows while the load is assumed to be constant (contigency analysis) or topology is constant (time series). 

Yes I coded these "quality of life" features, they have never been much used so maybe it's not perfect the way it is.

>  It would be nice if we could make these available in pp, too. However, in pp we have two additional steps beside the loadflow calculation: to translate the pandas tables to the ppc structure and the results back in the tables.

When using lightsim2grid, you have directly the possibility to get the result tables without having to convert to / from ppc and this is probably the main reason

> cient. It is probabily not easy to, for example, only translate new values for loads. However, I think the whole part can be implemented a lot faster using numba or c/c++.

This is exactly what I did in lightsim2grid: you have an "interface" to modify the loads, generators, shunts, sgens (and topology) and everything is converted to a Ybus, a Sbus and some pv, pq indices.

Then solvers use the Ybus, Sbus (and the pv, pq indices) to compute the powerflows. Recently i also implement the fact that you don't have to recompute the Ybus if topology do not change.

I can provide some benchmarks if you are interested in this broader than "just" allowing to chose between using lightsim2grid or not.

closed 2022-02-03 by rbolgaryn