e2nIEE/pandapower #1399: Inconsistent behaviour when multiple slacks on the grid (but you don't specify `distributed_slack=True` in the runpp function

BDonnot, 2021-11-17:
Hello,

I'm not sure if it's a bug or if i misunderstood something, but i came accross some strange behaviour when I tried to have a distributed slack bus on a grid.

Basically here is what I did:

1) I create a grid with no `ext_grid` (to reduce the likelyhood of an error) and with the column `net.gen["slack"] = True` for only one generator
2) launch a powerflow `pp.runpp(net)`, store the results (for the generators) in a dataframe
3) add a slack bus ( `net.gen["slack"] = True` for another generator by specifying correctly the slack_weight), relaunch a powerflow `pp.runpp(net)` and save the generator results in a dataframe
4) relaunch a powerflow, this time with `distributed_slack=True`:  `pp.runpp(net, distributed_slack=True)` and save the results likewise

I would have expected one of the two behaviours:

1) the grid acts like there is one single slack

    1) in case the "second generator" is taken as the slack bus, the results of case 3 should be identical of results of case 2
    2) in case the "first generator" is taken as the slack bus, the production active value (`p_mw`)of the other generator should not change between `gen` and `res_gen`
2) the grid acts like there is a distributed slack bus, in this case results of 3 and 4 above should match

Unfortunately, i got the results (code bellow):

```
Setpoint values
0     40.000000
1      0.000000
2      0.000000
3      0.000000
4    232.393272
Name: p_mw, dtype: float64
With only one slack (case 2)
0     40.000000
1      0.000000
2      0.000000
3      0.000000
4    236.412132
Name: p_mw, dtype: float64
Multiple slack, but nothing change in runpp (case 3)
0     40.000000
1    155.271171
2      0.000000
3      0.000000
4     69.875791
Name: p_mw, dtype: float64
Multiple slack, and distributed_slack=True (case 4)
0     40.000000
1      1.864340
2      0.000000
3      0.000000
4    234.257613
Name: p_mw, dtype: float64
```

So clearly, neither of what i could imagine is happening. I did not find anything detailing this in the documentation either.

The results of "case 3" are also super weird. Why a generator with a setpoint of 0. MW would see its generation increase up to 155MW and at the same time, the generator with a setpoint of 232 MW see its value decrease down to 69MW. This does not seem really appropriate.

Maybe one possibility would be to remove the "case 3" (or at least issuing a warning in this case) by checking that if `distributed_slack` is False and there are multiple "slack generators" on the grid, then there is "something unexpected".

Or maybe everything is consistent and I there is something I did not understand :-)

Thanks a lot

Benjamin

ps. The code to reproduce the results above:

```python
import pandapower as pp
import pandapower.networks as pn
import pdb
import copy

# retrieve the case14 and remove the "ext_grid" => put a generator as slack bus instead
case = pn.case14()
pp.runpp(case)  # forced to do it to retrieve the power of the slack bus
# adding a slack bus
var_pp = ["bus", 'in_service', "name", "max_p_mw", "min_p_mw", "max_q_mvar", "min_q_mvar", "slack_weight"]
slack_bus_gen_id_ppc = 0  # checked manually
pp.create_gen(case,
              **case.ext_grid[var_pp].iloc[0],
              p_mw=case._ppc['gen'][slack_bus_gen_id_ppc, 1],
              slack=True)
# "deactivate" the "ext_grid" 
case.ext_grid["in_service"][0] = False
pp.runpp(case)

# now create a copy of it, by removing the ext_grid completely (to be sure)
net = pp.create_empty_network("case14_custom", sn_mva=case.sn_mva, f_hz=case.f_hz)
# create bus
for i in range(case.bus.shape[0]):
    pp.create_bus(net, **case.bus.iloc[i])

# create lines
var_line = ["from_bus", "to_bus", "length_km", "r_ohm_per_km", "x_ohm_per_km", "c_nf_per_km", "max_i_ka", "name",
            "index", "type", "geodata", "in_service", "df", "parallel", "g_us_per_km",
            "max_loading_percent", "alpha", "temperature_degree_celsius"]
var_line = [el for el in var_line if el in case.line]
for i in range(case.line.shape[0]):
    pp.create_line_from_parameters(net, **case.line[var_line].iloc[i])

# create trafos
var_trafo = ["hv_bus", "lv_bus", "sn_mva", "vn_hv_kv", "vn_lv_kv", "vkr_percent", "vk_percent", "pfe_kw", "i0_percent", "shift_degree",
             "tap_side", "tap_neutral", "tap_max", "tap_min", "tap_step_percent", "tap_step_degree", 
             "tap_pos", "tap_phase_shifter", "in_service", "name", "index", "max_loading_percent",
             "parallel", "df"]
var_trafo = [el for el in var_trafo if el in case.trafo]
for i in range(case.trafo.shape[0]):
    pp.create_transformer_from_parameters(net, **case.trafo[var_trafo].iloc[i])

# create shunts
var_shunt = ["bus", "q_mvar", "p_mw", "vn_kv", "step", "max_step", "name",
             "in_service", "index"]
var_shunt = [el for el in var_shunt if el in case.shunt]
for i in range(case.shunt.shape[0]):
    pp.create_shunt(net, **case.shunt[var_shunt].iloc[i])

# create loads
var_load = ["bus", "p_mw", "q_mvar", "const_z_percent", "const_i_percent", "sn_mva", "name",
            "scaling", "index", "in_service", "type", "max_p_mw", "min_p_mw", "max_q_mvar",
            "min_q_mvar", "controllable"]
var_load = [el for el in var_load if el in case.load]
for i in range(case.load.shape[0]):
    pp.create_load(net, **case.load[var_load].iloc[i])

# create gens
var_gen = ["bus", "p_mw", "vm_pu", "sn_mva", "name", "index", "max_q_mvar", "min_q_mvar", "min_p_mw",
            "max_p_mw", "scaling", "type", "slack", "controllable", "vn_kv", "xdss_pu", "rdss_pu",
            "cos_phi", "in_service"]
var_gen = [el for el in var_gen if el in case.gen]
for i in range(case.gen.shape[0]):
    pp.create_gen(net, **case.gen[var_gen].iloc[i])

id_ref_slack = net.gen.shape[0]-1  # initial generator added as the slack bus added
net.gen["min_p_mw"][[id_ref_slack]] = 0.
net.gen["max_p_mw"][[id_ref_slack]] = 300.
assert (net.gen["min_p_mw"] >= 0.).all()
assert (net.gen["min_p_mw"] <= 400.).all()

# check the powerflow match
print("Setpoint values")
print(net.gen["p_mw"])
if True:
    print("With only one slack (case 2) ")
    pp.runpp(case)
    pp.runpp(net)

    assert ((case.res_gen - net.res_gen) <= 1e-6).all().all()
    assert ((case.res_load - net.res_load) <= 1e-6).all().all()
    assert ((case.res_shunt - net.res_shunt) <= 1e-6).all().all()
    assert ((case.res_line - net.res_line) <= 1e-6).all().all()
    assert ((case.res_trafo - net.res_trafo) <= 1e-6).all().all()
    assert ((case.res_bus.iloc[:net.bus.shape[0]] - net.res_bus) <= 1e-6).all().all()

    gen_without_single_slack = copy.deepcopy(net.res_gen)
    print(gen_without_single_slack["p_mw"])

# now activate more slack bus
net.gen["slack"][[1]] = True
net.gen["slack_weight"][[1, id_ref_slack]] = 0.5
pp.rundcpp(net)

# run the powerflow without distributed_slack=True
print("Multiple slack, but nothing change in runpp (case 3)")
pp.runpp(net, init="dc")
gen_without_dist_slack = copy.deepcopy(net.res_gen)
print(gen_without_dist_slack["p_mw"])

# run the powerflow with distributed_slack=True
print("Multiple slack, and distributed_slack=True (case 4)")
pp.runpp(net, init="dc", distributed_slack=True)
print(net.res_gen["p_mw"])
```

BDonnot, 2021-11-17:
> if there are 2 or more slack buses in the grid (2 gen, 2 ext_grid, 1 ext_grid 1 gen etc), and distributed_slack=False (default), the following behavior is expected: the load flow among them will be according to the power flow distribution in the grid.

I'm not sure i totally understands it. How are the power split on the grid ?

From my understanding (which may be incorrect), when computing a powerflow one single slack bus, the system has exactly as many unknowns than the number of equations (said differently: "you need a reference value for the voltage angle, setting it to 0 at the slack is the way to go").

If there are 2 slack buses (but you compute the powerflow "as if" there is only one), you "break" this symmetry (said differently: "you set to 0 the voltage angle at 2 buses") and at some point you will either not converge, or converge to something that does not represents anything (except in some corner cases where the theta would be zero at the two buses initially) because the voltage angle (hence the flows) are kind of "chosen by the user" for two different nodes.

For me again, the resolution with a distributed slack will "add" the "missing equation" (corresponding to the addition of the slack bus in the "single slack bus" case) in the form of a linear relation between the generators values.

I might be totally off in my understanding (which is possible) but I have a problem understanding how it would make sense to have a grid with 2 slack buses, but a powerflow without `distributed_slack=True`

> I expect that if a grid has 2 slacks (2 gens with slack=True), the resulting power values of gens are a) not necessarily equal b) not consistent with the slack_weight (because slack_weight is ignored if distributed_slack=False).

From my test, this appears to be correct indeed.

> If distributed_slack = True, then the resulting power values of gens will be consistent with slack_weight (according to the calculation formula in the tutorial).
> Does this describe the situation in your example?

I did not check with another powerflow, but it appears to be the case, yes. At least, the results I obtain with pandapower are consistent with this assertion, which is (for me) the correct behaviour to have.

rbolgaryn, 2021-11-22:
Dear @BDonnot ,

can you try connecting both slack buses with a switch and comparing the resulting bus voltages in the grid?
They should be equivalent to the situation 2 slack buses + distributed_slack=False

Best regards,
Roman

BDonnot, 2021-11-23:
Hello,

I'm not sure how to do this exactly. The code snippet I used is above and i'm sure it's pretty straigthforward to add these switches.

Benjamin

rbolgaryn, 2021-11-24:
Dear @BDonnot ,

the code below provides an example to illustrate the idea from my previous message. The results of the load flow calculation in a grid with 2 slacks without distributed_slack are equivalent to having just one slack bus that would be connected to the connection buses of the generators via a switch.

```python
import matplotlib.pyplot as plt

# first, obtain results with the case of 2 generators as slack
net.gen.loc[[1,4], 'slack'] = True
net.gen.loc[net.gen.slack, "vm_pu"] = 1.01
pp.runpp(net)

vm = net.res_bus.vm_pu.copy()
va = net.res_bus.va_degree.copy()

# now add a switch between the buses of the slack generators, and set only one generator as slack
net.gen.at[1, 'slack'] = False
pp.create_switch(net, 2,0, 'b')
pp.runpp(net)

fig, (ax1, ax2) = plt.subplots(1,2)
vm.plot(ax=ax1)
net.res_bus.vm_pu.plot(ax=ax1, linestyle='--')
va.plot(ax=ax2)
net.res_bus.va_degree.plot(ax=ax2, linestyle='--')
```

[image]

BDonnot, 2021-11-24:
Dear Romain,

Thank you for this experiment and detailed answer.

In your code it does as it put every slack bus at the same "theta" (by definition when not passing "distributed_slack=True" pandapower puts `theta = 0` for every slack buses) and you force all buses to have the same voltage magnitude "*to make a switch connection possible*". So you force all the voltages to be the same in all slack bus, which is exactly the behaviour of a bus/bus switch.

The analogy is far from perfect though. You can definitely run a powerflow with "distributed_slack=False" (and multiple slacks) and still have different voltages magnitude for each "slack bus". This behaviour cannot be obtained by adding a bus / bus switch.

So even if you look at the problem with the switch between all slack bus (which would be a really weird thing to do in my opinion), the actual behaviour is not really consistent (again still in my opinion)

rbolgaryn, 2021-11-27:
I will ask someone for a better explanation

lthurner, 2021-12-16:
> The results of "case 3" are also super weird. Why a generator with a setpoint of 0. MW would see its generation increase up to 155MW and at the same time, the generator with a setpoint of 232 MW see its value decrease down to 69MW. This does not seem really appropriate.

When a generator is set to slack, its active power is completely ignored. The power values are only determined by the electric resistances in the power grid.

BDonnot, 2021-12-17:
> When a generator is set to slack, its active power is completely ignored. The power values are only determined by the electric resistances in the power grid.

As far I understand powerflow sovlers, this is not completely true.

When running a single slack powerflow, you are totally correct. The "active power" of the slack is simply:
```
 P_slack = Sum_{load, load not slack} P_load + Sum_{line, trafo, ...} loss_{line, trafo, ...} - Sum_{gen, gen not slack} P_gen
 ```
 
 For a distributed slack, it's no more the case and the split is a bit more complicated.

All the generators are considered with their input value 'P_gen_target` , this would be the values in the `net.gen` and `net.sgen` from pandapower.

A (mathematical) hypothesis to run the powerflow is that, after the powerflow: `\sum P_gen = \sum P_load + \sum loss` (`P_gen` are the results values for generators, after the powerflow, the one in the table `net.res_gen`)

The way the distributed slack is implemented (see discussion #1408 especially the "slack" column of the jacobian matrix in the message https://github.com/e2nIEE/pandapower/issues/1408#issue-1066941484 ) is to "split" the difference `P_gen  - P_gen_target` according the to the "slack contribution factor" given in the "slack_weights" argument

So in this case, yes, `P_gen` is taken into account. You can check if you want. After a powerflow with a distributed slack is run, you can do: for all generator that contributes to the slack bus, compute `res_gen["p_mw"] - gen["p_mw"]`, sum the total values there (this will give exactly the power value absorbed by the slack), and then you can notice that `res_gen["p_mw"] - gen["p_mw"]` is proportional to the `slack_weights` column.

BDonnot, 2021-12-17:
To get back to the original remark I did, and maybe explain it differently.

To run a powerflow, you need a "slack bus" where you set the voltage angles to be 0. (because you need a *phase origin* -> basically, you say that you start to "count" when the voltage wave at this node is at 0.)

In case of a single slack bus, the `newton` function of pandapower does exactly that: it associate the voltage angle of the (unique in this case) slack bus and compute the others correctly.

In case of multiple node in the slack bus (so basically, you have more than one node that not `pv` not `pq`), the use of the same function will force **every** such buses to have a voltage angle of 0. (again you can check in pandapower using the `res_bus` table).
Though it is mathematically perfectly fine to do so, this computation does not mean anything "from the physical point of view". It would mean to define two distinct references for your system.

In some cases it's ok to do so that's the case in the example you provided above. In this example all buses participating to the distributed slack where at the same voltage magnitude, and you could connect them with switches. In this case, they are all represented with one bus (solver side) and it's perfectly fine (you only defined one "origin for the phase")

In every other cases, you get completely fictions flows from this computation. Basically the problem is not "what are the flows in my grid" but rather "what power do i need to inject here and there [*ie* in all buses participating to the slack bus] so that they have exactly the same voltage angles"

rbolgaryn, 2021-12-17:
`In every other cases, you get completely fictions flows from this computation. Basically the problem is not "what are the flows in my grid" but rather "what power do i need to inject here and there [ie in all buses participating to the slack bus] so that they have exactly the same voltage angles"`

I think this is relevant to the comment you made earlier, that if the voltage magnitude is not the same, the behavior is not consistent with the example of having multiple buses connected with a switch.

I think it is then possible to say that in the case if there are several slack buses defined in the grid, with different voltage setpoints, pandapower calculates what flows would be necessary to force the voltage magnitude and angle to match the setpoints.

rbolgaryn, 2021-12-17:
This would not be the only case such a situation can happen. For example, if there are several transformers in the grid, and they have different tap positions, maybe even with a different phase shift, you would also be able to observe flows in the grid that would be a nightmare for a grid operator if they happened in reality.

I don't know what's the best way to deal with the scenario when a grid has several slack buses that also have different voltage setpoints.

Maybe pandapower can include this as part of the pandapower diagnostic function and notify the user that such a configuration is suspicious. I think in the case when the slack buses have the same voltage setpoints, it is a matter of convenience of not having to create switches to connect the slack buses. In practice, I think having a distributed slack calculation is more reasonable then this configuration.

rbolgaryn, 2021-12-17:
In the case 3 with several slacks, and distributed_slack = False, the power values of the generators and slack_weight values are ignored.

If the calculation uses distributed_slack = True, the power values will be distributed among the generators as you wrote:

`The way the distributed slack is implemented (see discussion #1408 especially the "slack" column of the jacobian matrix in the message #1408 (comment) ) is to "split" the difference P_gen - P_gen_target according the to the "slack contribution factor" given in the "slack_weights" argument`

rbolgaryn, 2021-12-17:
Here you can see several examples for distributed slack in pandapower:

https://github.com/e2nIEE/pandapower/blob/develop/tutorials/distributed_slack.ipynb

BDonnot, 2021-12-20:
> I think it is then possible to say that in the case if there are several slack buses defined in the grid, with different voltage setpoints, pandapower calculates what flows would be necessary to force the voltage magnitude and angle to match the setpoints.

I think this is part of the issue I see. In this case, it does not compute "flows" to be necessary, but injections (P and Q) at the slack bus, where you force the voltage angle to be 0.

I'll try to say what I want to say differently. When you got a single slack, the unknowns at the slack bus are P and Q. The load-flow problem is to find P and Q such that you can get the right voltage angle (0) and the right voltage magnitude (given as input) at this bus.  This is fine when you have a single slack (closely related to "setting the origin of the voltage wave")

When you got a distributed slacks and you use the same algorithm, you will end up by fixing the P and Q at each of the distributed slack bus to make sure that the voltage magnitudes are set for all of theses buses as well as the voltage angles. As far as I know (but if that's a behaviour you'd like in pandapower, then no problem for me) this is not the goal of a powerflow.

One of the consequence of such is that you will totally ignore `slack_weights` in this case.

Another one is that you can totally set to 0 any flows you want: if a line connects bus `i` to bus `j` and both `i` and `j` are slack bus (with this definition) the active flow on this line will be 0. 

At the limit, when all buses participate to the distributed slack bus, you don't "compute flows" you compute the Sbus component so that all flows are 0. I'm not sure  again this is the right behaviour (but again, if that's the behaviour that you'd like then no problem with me) [and my intuition says that all active component of Sbus will be 0 in this case]

> I don't know what's the best way to deal with the scenario when a grid has several slack buses that also have different voltage setpoints.

It's already done in pandapower: the solution is to use a dedicated algorithm to deal with that, which translates, in pandapower, to set `distributed_slack=True`

In this case you use the right algorithm that allows you to 

> Here you can see several examples for distributed slack in pandapower:
Yes indeed, this was exactly what I was trying to say.

lthurner, 2022-01-31:
> In every other cases, you get completely fictions flows from this computation. Basically the problem is not "what are the flows in my grid" but rather "what power do i need to inject here and there [ie in all buses participating to the slack bus] so that they have exactly the same voltage angles"

I think you are completely right about this, but I disagree that this is the wrong behaviour. I think it is exactly the right behaviour.

If you specify two slack buses, you are defining the problem as: the voltage is forced to be at exactly this voltage (magnitude & angle) at two points in the system. Then pandapower calculates, how the current would flow through the network. pandapower does not give an opinion about if it is a good idea to operate a power grid like that, it calculates what would happen if you connect two perfect voltage sources.

If you connect two slack buses (=voltage sources) with a very low impedance, pandapower would calculate an extremely high current between these two slack buses. In the real world, that is called a short-circuit.

Another example from the real world: medium voltage networks are usually operated seperately (not connected galvanically), because if you would connect them, you would get large transit currents flowing from one HV/MV connection point to the next. That is because one HV/MV connection point will probably have a different voltage vector than the other HV/MV connection point. You can simulate this real world effect by connecting two slacks in pandapower. It will calculate the transit currents for you, that would ocurr in real life if you would galvanically couple the two MV networks.

So I believe 100% that pandapower is doing exactly what it is supposed to be doing with regards to calculating multiple slacks. This is also backed up by the fact that this also exactly how PowerFactory, SINCAL et. al model this situation. I don't see any need to change pandapowers behaviour here.

lthurner, 2022-01-31:
Maybe another example to make clearer what I mean. If you have two electric sockets at home, and connect them directly with a cable (DISCLAIMER: DON'T REALLY DO THAT), nothing would actually happen, if the voltages are exactly the same. This is your "connecting the same voltage slack forces the flow to be zero" situation. Nothing is forced here, it's just physics that there is no current flow if there is no voltage difference.

Now if the voltages fluctuate a little bit, and one socket has 230V and the second socket has 232V, you would get a small current through your connecting cable. If you would input the parameters into pandapower as two slacks with slightly different voltage magnitude and line with very small impedance, you could calculate this current flow.

I think you are thinking to deep about power flow algorithm and what your interpretation is about what they should calculate. In the end, we are trying to model the behaviour of an electrical system. This is what the electric system behaves like.

Now of course the distributed slack is also a valid model, it's just modelling a different situation. In this situation, your slack buses don't model a socket, but they model a large generating unit that can react to changes in the power system (mostly frequency) and adapt it's power output accordingly. To model such a behaviour, we need the distributed slack function.

So both behaviours are correct in my opinion, it's up to the user to decide what they want to model.

BDonnot, 2022-02-01:
Hello,

These were all suggestions. Feel free to adapt pandapower code and get the default behaviour you want.

I totally agree that for power system engineers, who understand the behaviour of the slack bus / distributed slack bus what pandapower is doing currently makes sense. People know what they are doing (hopefully) and thus get the right behaviour out of it.
And this is exactly why it's the default behaviour in commercial softwares.

However, for less experienced people with powergrids (for example people specializing in AI / ML / data science / etc. - one of the audience of pandapower if i'm not mistaken) and approaching a problem of powergrids, this behaviour is at best really confusing.
What is meant to be done by the user [distributing the slack on different buses] is not what is done in "one user action" [in this case adding a generator to the slack bus], the user also need to do `distributed_slack=True`.

One example: RTE provided powergrids in matpower, (*eg* `pn.case6495rte()`) with, of course, distributed slack that corresponds to a grid snapshot at a certain time (for example `2019-11-21 at 09:55` <- this is random)

When you load it, you have:
```
>>> net.ext_grid
    bus  in_service  name  va_degree  vm_pu  max_p_mw  min_p_mw  max_q_mvar  min_q_mvar
0  6077        True  None   8.095275  1.109   1387.00       0.0      416.10     -416.10
1  6161        True  None  17.012819  1.087   2898.84       0.0      869.65     -869.65
2  6305        True  None  15.044363  1.049   1503.00     403.0      657.68     -222.98
3  6306        True  None  15.133416  1.022   1503.00     403.0      657.68     -222.98
4  6307        True  None   0.515938  1.000   1498.00     398.0      390.29     -446.45
5  6308        True  None   0.525349  1.002   1498.00     398.0      368.81     -446.45
```
Notice how the voltage angles is different for all slacks buses.

If i put all the loads and generators of the French state at `2019-11-21 at 10:00` (so just after the one in `pn.case6495rte()`) and I run the powerflow (without specifying and extra key word argument), i will retrieve with pandapower a different value that the flows observed on the real grid (because pandapower has not modified the voltage angles of all the slack bus in this case).

That's apparently a known behaviour on your end so i'm closing the issue.

lthurner, 2022-02-01:
> What is meant to be done by the user [distributing the slack on different buses] is not what is done in "one user action" [in this case adding a generator to the slack bus], the user also need to do distributed_slack=True.

Yes, if you are modeling transmission systems. If the default behaviour would be distributed_slack=True, then it would be correct by default for transmission systems, but incorrect by default for distribution systems. We could of course implement a check that sets the default based on the voltage level, but I think that would be very implicit and unexpected behaviour, and in some systems like 110kV both use cases ocurr. In the end, there is no way around the user knowing what they are modeling. If you have a suggestion on how to elegently change the behaviour so that pandapower always uses the correct setting automatically, I am open to it. I just disagree that the current behaviour of pandapower is inconsistent or wrong, it's just meant for a different use case than yours.

A suggestion for the problem with the french grid: if you store it as a pandapower network, you can set the default calculation parameters with pp.set_user_pf_options. Then you can store it as a pandapower network and it will always be calculated with distributed_slack=True.

rbolgaryn, 2022-02-01:
I aslo thought about adjusting user_pf_options for all the rte grids, but @SteffenMeinecke suggested it is too implicit. 

I think it is better to show an info message with the suggestion to consider using distributed slack if the grid has multiple slacks.

BDonnot, 2022-02-01:
Yes the "solution" to add a default user option would make sense there.

Also, I think the difference is not "transmission" vs "distribution" but rather "single connex component" vs "multiple connex component".

If you have two slack buses in one single connex component, TSO or DSO it will not give "real" results but will suffer from the modeling artifact I described above.
It's the same if you don't specify a slack for a component of your grid (in case you have multiple subgrids): powerflow will not give meaningful results (it will most likely diverge in this case).

I totally agree that having this kind of checks will make the library less usable. Maybe issuing a warning to inform people that they have more than one slack for a connected component of the network and they probably want to have "distributed_slack=True" in this case.

lthurner, 2022-02-01:
I don't think its to implicit. Its a configuration of this network that allows you to calculate the network correctly by default. 

However, @BDonnot  was mentioning importing a matlab file. In that case, there is no way of changing the default behaviour without the user doing it after the import imho.

> I think it is better to show an info message with the suggestion to consider using distributed slack if the grid has multiple slacks.

But in that case it should only be shown if the slacks are actually galvanically connected. Having multiple slacks in one network that aren't connected is avery common use cases that shouldn't be given warnings for. It's just a question if that check is computationally efficient and really worth it just to show the warning message.

lthurner, 2022-02-01:
>If you have two slack buses in one single connex component, TSO or DSO it will not give "real" results but will suffer from the modeling artifact I described above.
It's the same if you don't specify a slack for a component of your grid (in case you have multiple subgrids): powerflow will not give meaningful results (it will most likely diverge in this case).

Again, the example you have given is for a transmission system, which everybody agrees its the wrong behaviour. But in a distribution system, it is actually the correct behaviour, as I have tried to explain in very much detail with the socket examples. It seems like you are not willing to accept this, but I don't quite understand why.

lthurner, 2022-02-01:
.... ok? seems like the discussion has run its course.

@rbolgaryn: i am all for changing the user parameter on the RTE networks in the pandapower networks package, seems like an easy an quick fix.

rbolgaryn, 2022-02-02:
> Hello,
>
> These were all suggestions. Feel free to adapt pandapower code and get the default behaviour you want.
>
> I totally agree that for power system engineers, who understand the behaviour of the [...]

Can you please elaborate a bit on how the distributed slack behavior is represented in the grid?

In MATPOWER, I do not see any extra variable that would describe the slack_weights. How would the slack weights be defined in the case6495rte?
I am assuming that the voltage angles at the ext_grid are result of some sort of calculation/scenario produced by RTE? 

Roman

BDonnot, 2022-02-02:
Hello,

I don't know how exactly these files have been generated. So i'm not sure what would be the slack weights. But you are correct, these have been generated by some process at RTE from SCADA / EMS measurments and then some anonymization procedure.

I don't know either how pandapower deduced from the matpower files what where the slack buses. I'm not familiar with that format.

If this data is not in matpower then i suppose you can have a uniform slack weights (all at 1.) or any other "heuristic"

Unfortunately, i cannot help you more

Benjamin

milindsmart, 2024-05-03:
I did not see a key point mentioned here.

The slack bus conventionally does two separate functions - sets a voltage angle reference, and covers losses in the network. For the first function it should be possible to specify the value of $`\delta`$. For the second function active power should be free to vary. So given the four parameters ($`P, Q, V, \delta `$) at each bus, the only possibilities of parameters that can be set are $`Q\delta`$ and $`V\delta`$ *. Since $`V\delta`$ specifies a voltage source, it is much preferred. 

But they don't have to have the same values for voltage and angle. In particular, the angle reference does not actually HAVE to be zero -  any number will work.

The physical and numerical significance of the slack bus is much more for the second function (covering the losses) than the first (setting the angle reference). Hence the name "slack" or "swing". 

In a multiple slack bus model (without distributed slack feature), the chosen slack buses should be free to be set independently at different voltage magnitude and angle values. Just specifying the angle at one bus doesn't change any physics but allows the solver to assign numerical phase values to all buses. Any further slack buses obviously change the flows to make the specified angles happen.

In a distributed slack bus model, a participation factor ("slack weight") is assigned to each bus, so that the mismatch power (losses generally) is soaked up in proportion to this factor. 

* Exactly two parameters have to be set at each bus, else it is no longer a powerflow problem, but something closer to an OPF where there parameters are free to be set.

BDonnot, 2024-05-03:
Hello,

Thanks for answering is issue. I don't remember exactly what was the problem (?) (and reading my first comment I was not even sure it was a problem at all) and don't have time to read this thread.

I think i'll close the issue, apparently you are confident that the behaviour of pandapower is perfectly correct (both in the case of converting matpower data to pandapower one and to the handling of multiple slack in non distributed slack setting) And you know pandapower much more than I do :-)

Feel free to re open it if I misjudge your previous message.

closed 2024-05-03 by BDonnot

milindsmart, 2024-05-03:
Yes you misjudged, but not the way you suspect perhaps! I do not know about pandapower itself - I was actually speaking purely from the theory of what the power flow problem means. Therefore, I was certainly not asserting that pandapower did the right thing. 

But perhaps you could have used the [`va_degree`](https://pandapower.readthedocs.io/en/latest/elements/bus.html#input-parameters) parameter to set a non-zero angle reference for your second "slack" bus, in the case where you did not want to use distributed_slack and the slack_weights. 

In your other issue #1408 you seem to have already grasped what I was saying earlier:

> (also, to account for multiple slacks, you put all slack buses as pv buses, except one, that you called the ref slack [I like the name, because that is the bus that has theta=0] and I will simply call "the slack" below)

Since I'm not a user of pandapower, I'll leave it to you and/others here about whether it should remain closed.