e2nIEE/pandapower #1404: passing multiple elements (as an array)  in a single controller

victoria274, 2021-11-23:
Hello I had an open issue about the timeseries calculation in my code, I took your advise to pass the element_index and profile_name like so

df=pd.read_csv("AUG_consumer_data_csv2.csv", index_col=0)
profiles=pd.DataFrame()
profiles=df["Weekdays kW"]
ds=DFData(profiles)

arraydata=np.array(profiles)   #passing array for profile_name

arrayload=np.array(net.load.p_mw)   #passing array for element_index

ConstControl(net, element='load', variable='p_mw', element_index=arraydata,
                data_source=ds, profile_name=arraydata)

ow = OutputWriter(net, time_steps=range(0,47), output_path ="./ts_out", output_file_type = ".csv")
ow.log_variable('res_load', 'p_mw')

pp.timeseries.run_time_series.run_timeseries(net, time_steps=48)

But this results in an error, not being able to run the timeseries

Another thing I have a question about is this note that I read in the documentation
"If multiple elements are represented with one controller, the data source must have integer columns. At the moment, only the DFData format is tested for the multiple const control." 

This means that all the data from the data source used must be integers, it will have an error if it has float numbers? Because from my data source csv file I have strings as the column's names and float as the data that is being read into the controller

Thank you

_Originally posted by @victoria274 in https://github.com/e2nIEE/pandapower/issues/1384#issuecomment-973236054_

rbolgaryn, 2021-11-24:
Dear @victoria274 ,

thank you for providing more details. 

I think it would be helpful to see the exact error message.

Also, I think the structure of the csv file and the dataframe should be checked.

ConstControl and DataSource expect that the index column represents a time step (int) and the dataframe has columns with the names for the profiles (to be used as the profile names with the parameter "profile_name"). The values in the columns of the datasource do not need to be integer.

You can verify if that is the case by executing: 

```python
print(net.controller.object.at[0].data_source.df)
```

Verify that the element_index is also correct, it should represent the element index from the pandapower element table (e.g. net.load.index.values). The length of element_index and profile_name must be the same. It should be okay if the profile_name has duplicated values (if you have just 1 column for data with profile name of "hello", but three loads to control, the parameter profile_name should be ["hello", "hello", "hello"]).

victoria274, 2021-12-03:
To answer some of your questions
1. I have attached the error message after running the above code is "Too many indexes" (see screenshots below)
2. The problem I think is the fact that the length of element_index and profile_name are not equal like you said. I have 27 loads and the timesteps are 48.
[image]
[image]

Thank you for your time, I will match the indexes of the two or try again with another looping method.

victoria274, 2021-12-05:
Hello again,
I still get an error even when I change the number of columns in my csv file to match the number of loads. Am I writing something wrong? 

df=pd.read_csv("AUG_consumer_data_csv3.csv", index_col=0)  
#this file has 28 columns, 1 index column for the timestamps and the other 27 for the load data 

df=pd.DataFrame(["Weekdays kW1", "Weekdays kW2", "Weekdays kW3", "Weekdays kW4", "Weekdays kW5", "Weekdays kW6", "Weekdays kW7", "Weekdays kW8", "Weekdays kW9", "Weekdays kW10", "Weekdays kW11", "Weekdays kW12","Weekdays kW13", "Weekdays kW14", "Weekdays kW15", "Weekdays kW16", "Weekdays kW17", "Weekdays kW18", "Weekdays kW19", "Weekdays kW20", "Weekdays kW21", "Weekdays kW22", "Weekdays kW23", "Weekdays kW24", "Weekdays kW25", "Weekdays kW26", "Weekdays kW27"])

ds=DFData(df)

ConstControl(net, element='load', variable='p_mw', element_index=net.load.index,
                data_source=ds, profile_name=["Weekdays kW1", "Weekdays kW2", "Weekdays kW3", "Weekdays kW4", "Weekdays 
                kW5", "Weekdays kW6", "Weekdays kW7", "Weekdays kW8", "Weekdays kW9", "Weekdays kW10", "Weekdays kW11", 
               "Weekdays kW12","Weekdays kW13", "Weekdays kW14", "Weekdays kW15", "Weekdays kW16", "Weekdays kW17", 
               "Weekdays kW18", "Weekdays kW19","Weekdays kW20", "Weekdays kW21", "Weekdays kW22", "Weekdays kW23", 
               "Weekdays kW24", "Weekdays kW25", "Weekdays kW26", "Weekdays kW27"])

ow = OutputWriter(net, time_steps=48, output_path ="./ts_out", output_file_type = ".csv")
ow.log_variable('res_load', 'p_mw')

pp.timeseries.run_time_series.run_timeseries(net, time_steps=48)

[AUG_consumer_data_csv3.csv](https://github.com/e2nIEE/pandapower/files/7655760/AUG_consumer_data_csv3.csv)

rbolgaryn, 2021-12-10:
Hi @victoria274 ,

key error is showing that you have a list with 1 object inside, namely the Index([....]).

instead, you need to provide df.columns.values to profile_name, or the list of column names exactly as you wrote in the code portion with ConstControl.

victoria274, 2021-12-17:
I changed the code with your feedback and it runs halfway this time but still gets an error
"ValueError: Must have equal len keys and value when setting with an iterable"

df=pd.read_csv("AUG_consumer_data_csv3.csv", index_col=0)
ds=DFData(df)

ConstControl(net, "load", "p_mw", element_index=net.load.index, profile_name=df.columns.values, data_source=ds)

ow=OutputWriter(net, time_steps=(0,48), output_path="results", output_file_type=".csv")
ow.log_variable("res_load", "p_mw")

run_timeseries(net, time_steps=(0,48))

rbolgaryn, 2021-12-17:
Hi @victoria274 ,

time_steps should be range(0,48) or range(48), and same in run_timeseries.

But it is probably separate from the issue you're having right now.
I think the only chance to figure it out is if you debug the code and catch the root cause of the error. 

Roman

victoria274, 2021-12-17:
Hi @rbolgaryn,
I changed it but it gives me this error KeyError: 0

rbolgaryn, 2021-12-17:
Sorry I don't think I can be helpful

victoria274, 2021-12-17:
thank you for your time Roman :)

closed 2021-12-22 by rbolgaryn