e2nIEE/pandapower #1408: Maybe some issues with distributed slack Newton Raphson

BDonnot, 2021-11-30:
Hello,

I dived a bit (maybe too deep though) in the Newton Raphson algorithm to compute powerflow, especially for the "distributed slack" implementation and I think there is a "bug" there, or rather a discrepency in the computation of the Jacobian matrix. I'm not 100% sure, but I still think there should be a rewriting of some of the functions (`newtonpf`, `_evaluate_Fx` and `create_J_ds`) there.

# What I understood

The way I understood the current implementation is that , compared to the single slack implementation, you add an extra dimension in the Jacobian, one row to accompt for the slack weights (how to "split the losses" between the slack buses) and one equation that accounts for the "reference slack". Adding one equation and one variable is perfectly fine.

(also, to account for multiple slacks, you put all slack buses as pv buses, except one, that you called the `ref slack` [I like the name, because that is the bus that has `theta=0`] and I will simply call "the slack" below)

# What I noticed

However, i think, in the building of the Jacobian, there are some discrepencies (i am looking at the `pandapower/pypower/newtonpf.py` file). 

The column added is the last one which is fine, and it's properly implemented in :
```python
        if dist_slack:
            slack = slack + dx[j7:j8]  # j7 and j8 always point to the final component of dx
```
However, if we now look at the `_evaluate_Fx` function, we can notice that the last index of the results if in fact the imaginary mismatch of the last pq bus:
```python
F = r_[mis[ref].real, mis[pv].real, mis[pq].real, mis[pq].imag]
```

Maybe that's me, but I don't see how the last component of the pq bus is equal to the slack bus.

For me, if we look at the `newtonpf` function, then the `mis[ref].real` should be the last component, because it indeed represents the slack.

# What I can suggest

For me, i think there are two possible options.

Option 1: you put the slack weights as the first columns of the Jacobian matrix (instead of the last column as of now), keep slack bus equation as the first row, and adjust the way you update Va and Vm and slack:
```
        # update voltage
        if npv and not iwamoto:
            Va[pv] = Va[pv] + dx[j1:j2]  # have j1 starts at 1 and nex j2 be old j2 +1
        if npq and not iwamoto:
            Va[pq] = Va[pq] + dx[j3:j4]  # new j3 is old j3 +1 (same for j4)
            Vm[pq] = Vm[pq] + dx[j5:j6] # new j5 should be old j5 +1
        if dist_slack:
            slack = slack + dx[j7:j8]  # j7 should be 0 and j8 = 1
```

Option 2: you keep the slack variable as the last column of the Jacobian, put the slack  equation at the last row of the Jacobian, don't change anything to the `newtonpf` function but change the `_evaluate_Fx` consequently (to change the fact that now the slack variable is indeed the last one):

```python
F = r_[mis[pv].real, mis[pq].real, mis[pq].imag, mis[ref].real]
```

In both options, the function `create_J_ds` (in `pandapower\pf\create_jacobian_numba.py`) should be adapted to reflect the changes.

Currently we have:

```
    | slack_bus | s |              |    (1,pvpq)   (1, pq)     |(pvpq+1,1) |
    |  -------  | l |              | ------------------------- |           |
    | J11 | J12 | a |              | (pvpq, pvpq) | (pvpq, pq) |           |
    | --------- | c |= dimensions: | ------------------------- |   -----   |
    | J21 | J22 | k |              |  (pq, pvpq)  | (pq, pq)   |  (pq, 1)  |

    `J11` = dS_dVa[array([pvpq]).T, pvpq].real
    `J12` = dS_dVm[array([pvpq]).T, pq].real
    `J21` = dS_dVa[array([pq]).T, pvpq].imag
    `J22` = dS_dVm[array([pq]).T, pq].imag

    `slack_bus` = is the representation of the equation for the slack bus dS_dVa[slack_bus_id, pvpq].real
    and dS_dVm[slack_bus_id, pq].real

    `slack` is the representation of the equation connecting together the slack buses (represented by slack_weights)
    the remaining pq components are all 0.

```
In option 1 above, taking the notation above, we would have
```
    | s | slack_bus |               | (pvpq+1,1) |    (1,pvpq)   (1, pq)     |
    | l |  -------  |               |            | ------------------------- | 
    | a | J11 | J12 |               |            | (pvpq, pvpq) | (pvpq, pq) |
    | c | --------- | = dimensions: |   -----    | ------------------------- |
    | k | J21 | J22 |               |  (pq, 1)   |  (pq, pvpq)  | (pq, pq)   |
```

In option 2 above, taking the notation above, we would have
```
                
    | J11 | J12 | s |              | (pvpq, pvpq) | (pvpq, pq) | (pvpq,1) |
    | --------- | l |= dimensions: | ------------------------- |   -----  |
    | J21 | J22 | a |              |  (pq, pvpq)  | (pq, pq)   | (pq, 1)  |
    |  -------  | c |              | ------------------------- |   -----  |
    | slack_bus | k |              |    (1,pvpq)   (1, pq)     |    sw    |

# with sw being the component `slack_weights[ref]` (ie the weights associated to the reference slack bus) 
```

rbolgaryn, 2022-01-31:
Dear @BDonnot ,

I think for the Option 1, which you implemented in lightsim2grid, the order in which the slack value should be integrated in the F and dx are:

* first position in the F (mismatch vector should start with the mismatch of P at the ref bus)
* first position in the dx (slack power)

Do you agree?

BDonnot, 2022-01-31:
Hello,

Indeed I implemented option 1 in lightsim2grid.

The first position of the result of `evaluate_Fx` is the real power mismatch at the ref slack bus `real_(slack_id)`, and I update the `slack` variable with the rule `slack_absorbed -= F(0)`.

rbolgaryn, 2022-01-31:
Thank you!

I will try modifying the J in pandapower and then compare it with J in lightsim2grid.

closed 2022-02-03 by rbolgaryn