e2nIEE/pandapower #1409: Better "initial guess for the slack"

BDonnot, 2021-11-30:
Hello, 

A smaller "issue" (that is not really one) I saw in the `newtonpf` implementation concerned the initialisation of the slack bus.

Indeed, when `sn_mva` is not 1. initializing `slack = gen[:, PG].sum() - bus[:, PD].sum()` might completely overestimate / underestimate (depending on whether sn_mva > 1 or not) the power absorbed by the slack bus initially.

A possibility would be to  either initialize it with `slack = (gen[:, PG].sum() - bus[:, PD].sum()) / baseMVA ` or to intialize it with  `slack = np.real(np.sum(Sbus))` (which is another initialization, it will not give the same value as the targeted one)

rbolgaryn, 2021-12-02:
Hi @BDonnot ,

thank you! The first option looks the best to me, can you please make a pull request for it?

Roman

rbolgaryn, 2021-12-03:
I have opened the PR

BDonnot, 2021-12-03:
I did not had time to make it, thanks :-)

Do you want me to do anything there ?

Benjamin

rbolgaryn, 2021-12-03:
No problem, I think it is complete. Thank you for the suggestion!

closed 2021-12-03 by rbolgaryn