e2nIEE/pandapower #1410: IO with json misses dataframes index name

SteffenMeinecke, 2021-11-30:
If a pass a net with information `net[element].index.name` to `to_json()` and load the net again with `from_json()`, the information is about the index name is lost (already with `to_json()`). It would be a nice feature to include the data to the fileIO if this can be done easily.

Here a test script to highlight the issue:

```
import pandapower as pp
import pandapower.networks as pn
net = pn.simple_four_bus_system()
bus_id = "bus_id"
net.bus.index.name = bus_id
file = "test_net.json"
pp.to_json(net, file)
net = pp.from_json(file)
print("The following print should show '%s':" % bus_id)
print(net.bus.index.name)
```

dlohmeier, 2021-12-01:
This is a problem with the serialization method using pandas' `to_json` method. This doesn't work:

    import pandas as pd
    from pandas.testing import assert_frame_equal
    df = pd.DataFrame({"a": [1, 2], "b": [2, 4]})
    df.index.name = "hello"
    s = df.to_json()
    df2 = pd.read_json(s)
    assert_frame_equal(df, df2)

However, this works:

    import pandas as pd
    from pandas.testing import assert_frame_equal
    df = pd.DataFrame({"a": [1, 2], "b": [2, 4]})
    df.index.name = "hello"
    s = df.to_json(orient="table")
    df2 = pd.read_json(s, orient="table")
    assert_frame_equal(df, df2)

If required, we could add an option of how pandas dataframes are stored. Currently, this is done with the "split" orientation, but this could be changed. However, if we add this option to our `to_json` function, we should decide to store this metadata as well, as this data is lost within pandas. Otherwise a user must know that this option was chosen when storing the net. I believe that this is not a big problem, but requires some effort.

SteffenMeinecke, 2021-12-01:
@dlohmeier thanks for elaborating this issue. Since this feature is not mandatory for my workflows and pandapower doesn't use `net[elm].index.name` we can close this issue for now. If this issue come up again later, we can address the proposed implementation. Or we are lucky and this issue disappear with future changes of pandas.

closed 2021-12-01 by SteffenMeinecke