e2nIEE/pandapower #1414: Problem with JSON serialization / deserialization when using numpy dtypes

dlohmeier, 2021-12-03:
In PR #1413 I came across a problem in one of the tests that makes use of a Tap Controller. In this controller type, some numpy types are used for attribute values, e.g. the attribute `element_in_service` is of type numpy._bool instead of bool. This can lead to errors in the JSON serialization / deserialization:

- In case of boolean values, the value is saved as string, when the type is numpy._bool. It is then retrieved as string and cannot be used correctly any more
- In case of numerical (e.g. numpy.float64), the precision between the original and retrieved value can differ, when restoring it as float.

How can we overcome this problem? This probably only occurs for JSONSerializables or pandapowerNet attributes, as we don't save other types of objects (except numpy and pandas, which automatically deal with this). I would suggest to change the serialization of numpy dtypes to also use signatures.

@rbolgaryn @lthurner Can you comment on this?

dlohmeier, 2021-12-03:
I tried to make this work, however I realized that the decoder registry for numpy.floating is never used, as there is a float str representation by default implemented in the PPJSONDecoder. We should have a look at it and rather override the parts we are sure that they work with our registry.

dlohmeier, 2021-12-03:
The problem with the floating type can be circumvented by adding numpy.floating to the `with_signature` function given to the Encoder.

rbolgaryn, 2021-12-03:
The method floatstr that is defined inside the method iterencode is just a copy-paste from the JSONdecoder (base class).

It seems like the json package is always falling back on the floatstr to save floating numbers by default.

PR #1416 (2021-12-03): save numpy objects with signature in JSON encoder 
closes #1414
closed 2021-12-03 by rbolgaryn