e2nIEE/pandapower #1419: PTDF/LODF Calculation

thomsonian2023, 2021-12-04:
I have a PSSE RAW file and I am looking to evaluate PTDF and LODF. Is this possible with pandapower ?

rbolgaryn, 2021-12-13:
dear @thomsonian2023 ,

it should be possible to use the functions 

* pandapower.pypower.makePTDF.makePTDF() 
* pandapower.pypower.makeLODF.makeLODF() 

Roman

jhingu, 2021-12-13:
Hi Roman, what function should I apply to PSSE/RAW file to create input format for pandapower.pypower.makePTDF.makePTDF()

rbolgaryn, 2021-12-13:
Hi @jhingu ,

we don't provide the converter from PSSE/RAW file format.

You can import the grid data in one of the following ways:

* saving it to MATPOWER mpc format first and then importing the grid from MATPOWER format
* saving all the relevant data to an Excel file (in the format that pandapower expects) and importing the grid data from excel with pp.from_excel
* write a converter function from PSEE (and contribute it to pandapower)

Please let us know which option you use.

Roman

closed 2022-01-14 by rbolgaryn