e2nIEE/pandapower #1425: Implement transformer tap changer controller with Line Drop Compensation (LDC)

rbolgaryn, 2021-12-09:
* Controller for OLTC transformer (modification of TrafoControl?)
* Controlling the voltage at a remote bus
* Parameter for the impedance of a fictitious line between the transformer and the remote bus (optional: calculate based on topological search)
* Idea: use the current to calculate the expected voltage drop and compensate it with OLTC

This strategy is sometimes used in radial distribution grids. Maybe not so useful in grids with multiple feeders and high distributed generation. Potentially useful in simple grids with just 1 feeder.

pawellytaev, 2024-11-26:
@SteffenMeinecke @vogt31337 does this sound like a useful feature to you? what do you think?

SteffenMeinecke, 2024-11-26:
> This strategy is sometimes used in radial distribution grids.
Obviously @rbolgaryn had to deal with an use case where this kind of controller could or should have been used. I agree with @rbolgaryn that this would be a perfect issue that could be added by someone from the open souce community.