e2nIEE/pandapower #1430: BUG in rerun a powerflow with dclines since v2.6.0

SteffenMeinecke, 2021-12-15:
It is noticed by the bug report https://github.com/e2nIEE/simbench/issues/20, that rerunning a powerflow, e.g. by using the timeseries module, with a net including dclines raises an error. This is caused due to the internally created additional generators (https://github.com/e2nIEE/pandapower/blob/develop/pandapower/auxiliary.py#L908) which then exist in `net["_is_elements"]["gen"]` but does not exist in the bus_lookup at https://github.com/e2nIEE/pandapower/blob/develop/pandapower/build_gen.py#L210. This error first occure with v2.6.0. I didn't check which commit is the critical one.

Here an code snippet to reproduce the error:

```
import pandapower as pp
from pandapower.timeseries.run_time_series import run_timeseries
import simbench as sb

what_to_test = "both"  # "runpp", "ts", "both"

grid_code = "1-EHVHV-mixed-all-1-no_sw"
net = sb.get_simbench_net(grid_code)

profiles_non_working = sb.get_absolute_values(
    net, profiles_instead_of_study_cases=True)

sb.apply_const_controllers(net, profiles_non_working)

assert not sb.profiles_are_missing(net)

timesteps = range(2)

if what_to_test == "runpp":
    pp.runpp(net)
    # no error
elif what_to_test == "ts":
    run_timeseries(net, timesteps)
    # error at 2nd time_step
elif what_to_test == "both":
    pp.runpp(net)
    run_timeseries(net, timesteps)
    # error at 1st time_step
```

SteffenMeinecke, 2022-02-05:
When this bug is fixed once, a test should be added to ensure the functionality.

KS-HTK, 2025-11-27:
There seems to be a different issue here.

I get `pandapower.auxiliary.LoadflowNotConverged: Power Flow nr did not converge after 10 iterations!` even for the runpp case.