e2nIEE/pandapower #1434: WARNING: pandapower.estimation.state_estimation:Estimation failed! Pandapower network failed to update!

PenelopeR, 2021-12-21:
Does anybody understand what this warning means? This happens when we run success = estimation.estimate(net, init='flat') for state estimation. The success of the state estimation after this warning = False. We have a net of 16 buses of which 12  have measurement data. Using pseudo measurements we make the net observable. Occasionally the state estimation runs fine (success = True) and at other times it gives the above warning and success = False. When I run pandapower.diagnostic(net, report_style='detailed) all tests are passed. I have also made sure the units of all the inputs are accurate. A bit at a loss what this warning means.

wangzhenassd, 2021-12-22:
This information indicates the state estimation failed similar to the non-convergence by loadflow calculation

PenelopeR, 2021-12-23:
Thank you. This allows me to debug accordingly :) It probably has to do with the network observability assumption we are making based on the rank.

closed 2021-12-23 by PenelopeR