e2nIEE/pandapower #1435: Error in transformer documentation

senzhanopt, 2021-12-22:
On the transformer section, regarding the definition of transformer capacity-based per-unit system, the impedance base value is defined as 

<img src="https://render.githubusercontent.com/render/math?math=Z_{ref,trafo} = \frac{vn\_lv\_kv^2 \cdot net.sn\_{mva}}{sn\_{mva}}">

I think the extra <img src="https://render.githubusercontent.com/render/math?math=net.sn\_mva"> is not needed.

https://pandapower.readthedocs.io/en/develop/elements/trafo.html

SteffenMeinecke, 2022-01-10:
Dear @senzhanopt,
thanks for raising your issues here.
In that particular case, I think we need to keep `net.sn_mva` in the documentation. See https://github.com/e2nIEE/pandapower/issues/866 for the discussion why it was added. You can also have a look to the code https://github.com/e2nIEE/pandapower/blob/develop/pandapower/build_branch.py#L495 - `net.sn_mva` is considered here as `sn_mva` to calculte `tap_lv`, which is used for `z` and `r` as stated in the documentation

rbolgaryn, 2022-01-11:
I think it would help to state in the documentation, which variables are in p.u. and which are absolute

closed 2022-02-02 by senzhanopt