e2nIEE/pandapower #1436: The problem of node label mismatch caused by the wrong format of geographic data.

roki1031, 2021-12-24:
Hi~
     On the journey of learning pandapower, I found that the Bus id in the grid structure diagram drawn by the function `plot.create_annotation_collection` does not seem to match the actual bus.
[image]
[image]
     As you can see, the beginning and end nodes of line 1 should be bus 0 and bus 4, but in the grid diagram they are bus 0 and bus 8.

     So i digged into the source code and found that it was caused by the wrong format of geographic data.

     After running this line of code, the grid diagram is correct. `net.bus_geodata.sort_index(inplace=True)`
     
[image]

rbolgaryn, 2022-01-07:
Thank you for finding this problem, @roki1031 !

I think the solution should not be sort_index, for the case if the bus indices in net.bus are not sorted. I think a better solution would be to use the indices of net.bus when using net.bus_geodata:

net.bus_geodata.loc[net.bus.index]

Best regards,
Roman

rbolgaryn, 2022-01-14:
I looked at the source code of create_annotation_collection, and it just takes texts and coords as arguments.

So it is entirely the responsibility of the user to make sure that the order of texts and coordinates match.

Roman

closed 2022-01-14 by rbolgaryn