e2nIEE/pandapower #1440: from_mpc fail

tyua321, 2021-12-30:
The mpc file I generated from MATPOWER can not run in Pandapower.

I tried to generate case14 file from MATPOWER

```
mpc = pc.from_mpc('mpc.mat',f_hz=60)
pp.runpp(mpc)
```
there are some warnings in `from_mpc`
```
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\converter\pypower\from_ppc.py:222: RuntimeWarning: invalid value encountered in double_scalars
  max_i_ka = ppc['branch'][i, 5]/ppc['bus'][to_bus, 9]/sqrt(3)
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\converter\pypower\from_ppc.py:230: RuntimeWarning: divide by zero encountered in double_scalars
  c_nf_per_km=ppc['branch'][i, 4]/Zni/omega*1e9/2,
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\converter\pypower\from_ppc.py:230: RuntimeWarning: invalid value encountered in double_scalars
  c_nf_per_km=ppc['branch'][i, 4]/Zni/omega*1e9/2,
```
and when I ran power flow on it, it failed
```
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:138: RuntimeWarning: invalid value encountered in true_divide
  branch[f:t, BR_R] = line["r_ohm_per_km"].values * length_km / baseR / parallel
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:139: RuntimeWarning: invalid value encountered in true_divide
  branch[f:t, BR_X] = line["x_ohm_per_km"].values * length_km / baseR / parallel
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:150: RuntimeWarning: invalid value encountered in multiply
  b = 2 * net.f_hz * math.pi * line["c_nf_per_km"].values * 1e-9 * baseR * length_km * parallel
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:386: RuntimeWarning: invalid value encountered in true_divide
  (u1 + du * cos(tap_angles))))
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:441: RuntimeWarning: invalid value encountered in true_divide
  tap_rat = vn_hv_kv / vn_lv_kv
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:445: RuntimeWarning: invalid value encountered in true_divide
  get_values(ppc["bus"][:, BASE_KV], lv_bus, bus_lookup)
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:416: RuntimeWarning: invalid value encountered in true_divide
  np.square(vn_trafo_lv / vn_lv) * sn_mva  # adjust for low voltage side voltage converter
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:323: RuntimeWarning: invalid value encountered in true_divide
  b_real = pfe / vnl_squared * baseR
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:329: RuntimeWarning: invalid value encountered in true_divide
  b_img = np.sqrt(b_img) * baseR / vnl_squared
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:331: RuntimeWarning: invalid value encountered in true_divide
  return y / np.square(vn_trafo_lv / vn_lv_kv) * parallel
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:291: RuntimeWarning: invalid value encountered in true_divide
  zc_star = -1j / y[tidx]
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:293: RuntimeWarning: invalid value encountered in true_divide
  zab_triangle = zSum_triangle / zc_star
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:294: RuntimeWarning: invalid value encountered in true_divide
  zbc_triangle = zSum_triangle / za_star
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_branch.py:297: RuntimeWarning: invalid value encountered in true_divide
  y[tidx] = -2j / zbc_triangle
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\build_bus.py:463: RuntimeWarning: invalid value encountered in true_divide
  v_ratio = (ppc["bus"][bus_lookup[s["bus"].values], BASE_KV] / s["vn_kv"].values) ** 2
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\pypower\makeYbus.py:80: RuntimeWarning: invalid value encountered in true_divide
  Ysf = stat / (branch[:, BR_R] + 1j * branch[:, BR_X])  ## series admittance
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\pypower\makeYbus.py:93: RuntimeWarning: invalid value encountered in true_divide
  Yff = (Ysf + 1j * Bc / 2) / (tap * conj(tap))
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\pypower\makeYbus.py:94: RuntimeWarning: invalid value encountered in true_divide
  Yft = - Ysf / conj(tap)
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\pypower\makeYbus.py:95: RuntimeWarning: invalid value encountered in true_divide
  Ytf = - Yst / tap
C:\Users\81568\Anaconda3\lib\site-packages\scipy\sparse\linalg\dsolve\linsolve.py:206: MatrixRankWarning: Matrix is exactly singular
  warn("Matrix is exactly singular", MatrixRankWarning)
C:\Users\81568\Anaconda3\lib\site-packages\pandapower\pf\create_jacobian.py:18: RuntimeWarning: invalid value encountered in true_divide
  dVm_x, dVa_x = dSbus_dV_numba_sparse(Ybus.data, Ybus.indptr, Ybus.indices, V, V / abs(V), Ibus)
Traceback (most recent call last):

  File "<ipython-input-440-2692533814bf>", line 1, in <module>
    pp.runpp(mpc)

  File "C:\Users\81568\Anaconda3\lib\site-packages\pandapower\run.py", line 225, in runpp
    _powerflow(net, **kwargs)

  File "C:\Users\81568\Anaconda3\lib\site-packages\pandapower\powerflow.py", line 77, in _powerflow
    _ppci_to_net(result, net)

  File "C:\Users\81568\Anaconda3\lib\site-packages\pandapower\powerflow.py", line 178, in _ppci_to_net
    "{1} iterations!".format(algorithm, max_iteration))

LoadflowNotConverged: Power Flow nr did not converge after 10 iterations!
```

However the power flow on 30 bus system in pandapower is OK.
```
mpc30= pc.from_mpc('mpc30.mat', casename_mpc_file='mpc30',f_hz=50)
pp.runpp(mpc30)
```
there is a discrepancy between the `case14()` in pandapower and the case file from MATPOWER
the mpc loaded from MATPOWER

```
   - bus (14 elements)
   - load (11 elements)
   - gen (4 elements)
   - shunt (1 element)
   - ext_grid (1 element)
   - line (17 elements)
   - trafo (3 elements)
   - poly_cost (5 elements)
```

and `case14()`

```
   - bus (14 elements)
   - load (11 elements)
   - gen (4 elements)
   - shunt (1 element)
   - ext_grid (1 element)
   - line (15 elements)
   - trafo (5 elements)
   - poly_cost (5 elements)
   - bus_geodata (14 elements)
```

I think there is something wrong on the branch data when I loaded mpc file.
can you help me with it?

rbolgaryn, 2022-01-07:
Hi @tyua321 ,

can you check if any parameters in net.line are NaN?

Roman

rbolgaryn, 2022-01-07:
It could also be easier for you to use the "case" models from pandapower.networks:

https://pandapower.readthedocs.io/en/v2.6.0/networks/power_system_test_cases.html#case-14

SteffenMeinecke, 2022-01-10:
As @rbolgaryn linked, the `baseKV` had been filled by data from the linked source. The converter cannot calculate branch parameters with `baseKV=0`.

closed 2022-01-14 by rbolgaryn