e2nIEE/pandapower #1442: Importing OpenDSS files

eddie-atkinson, 2022-01-05:
Hi, 

I am interested in using Pandapower to model some of the feeders in the Australian [National Low-Voltage Feeder Taxonomy Study](https://near.csiro.au/assets/f325fb3c-2dcd-410c-97a8-e55dc68b8064). However, all of the files are in the format used by OpenDSS. 

 The [OpenDSSDirect.py](https://dss-extensions.org/OpenDSSDirect.py/notebooks/GettingStarted.html) package appears to offer the ability to export OpenDSS models as Pandas dataframes. I imagine the  process would be to convert the lines, loads, transformers and buses to dataframes and then to transform the relevant parameters to their pandapower equivalent.

However, I have no experience with OpenDSS, and am unsure what all the parameters mean.

Has anyone else successfully used OpenDSS models in Pandapower?

eddie-atkinson, 2022-01-07:
I've been doing a bit of reading to figure out if this is possible and I thought I'd summarise my thoughts here for the benefit of others:

1. It is possible to convert OpenDSS models to Pandapower, but there is no way to do it in an automated fashion. The best approach would be to look at the `.dss` files and translate them to Pandapower by hand. 
2. Whether or not it's possible to convert an OpenDSS model to Pandapower depends on the kind of modelling being performed. If you need to do some unbalanced power flow modelling you're out of luck.
3. It would likely be possible to convert *some* of the National Low Voltage Feeder Taxonomy models to Pandapower, however the ones I care about (where significant DER penetration causes voltage instability) have single phase loads which I don't think I can model in Pandapower. This is because Pandapower is primarily focused on modelling European grids, where single phase loads are uncommon. 

I will keep this issue open in case any of the Pandapower maintainers wish to correct me.

rbolgaryn, 2022-01-07:
Dear @eddie-atkinson ,

thank you!

If it is possible for you to convert the grids manually, I think it should be possible for you to write a script that would translate the OpenDSS format to pandapower format.

In pandapower it is possible to model 1ph loads and run a 3ph unbalanced powerflow. However, only the grids that have a symmetric feeder design are supported (3-phase with or without neutral with same rated voltage).

We are currently working with other partners to implement their algorithm of multi-phase power flow calculation that would enable calculating power systems with unsymmetric feeder design (e.g. USA-style, SWER, etc.). We plan to introduce this set of features later this year (probably 2H).

Roman

eddie-atkinson, 2022-01-07:
Hi @rbolgaryn , 

Thanks for responding and thank you for your work on this awesome package. 

> In pandapower it is possible to model 1ph loads and run a 3ph unbalanced powerflow
How do I model the single phase load? Do I just use an asymmetric load with load on a single phase and no load on the other phases? 

Cheers, 

Eddie Atkinson

rbolgaryn, 2022-01-07:
You are welcome!

You can use the asymmetric load and asymmetric sgen elements, which have inputs for phases a, b, c.

You also need to add zero sequence parameters to lines and transformers, and specify the vector group for transformers and set wye/delta types to relevant elements.

https://pandapower.readthedocs.io/en/v2.6.0/powerflow/ac_3ph.html

Best regards,
Roman

rbolgaryn, 2022-01-07:
I recommend using develop branch

closed 2022-01-14 by rbolgaryn