e2nIEE/pandapower #1449: PandaPower install fails for Python 3.10

maurerle, 2022-01-14:
The Scipy 1.6 wheel for Python 3.10 does not exist.
The version requirement for `scipy<=1.6.0` should be removed as scipy 1.7 works fine with 3.10 but previous versions don't.

I don't know why it was added here: https://github.com/e2nIEE/pandapower/commit/c06a1c22a54dc4d1691d0645adafbf1976e10fe2

rbolgaryn, 2022-01-14:
This is the reason:

https://github.com/e2nIEE/pandapower/issues/1146

rbolgaryn, 2022-01-14:
We need to find the time to fix it, before we can have scipy 1.7

maurerle, 2022-01-14:
Wouldn't pinning numpy to < 1.20 and leaving scipy without version requirement help?

Otherwise the referenced issue is closed since 4 days on develop, so I hope the fix will come in a few weeks :)

rbolgaryn, 2022-02-02:
3.10 not possible yet because fiona is not supporting 3.10 yet (should support with the next release).

closed 2022-02-08 by rbolgaryn