e2nIEE/pandapower #1451: remove_bad_data fails: A problem appeared while using the linear algebra methods.Check and change the measurement set.

jnsbal, 2022-01-19:
Hello,

I am currently experimenting with pandapower and experiencing an issue that is hard for me to troubleshoot.
Basically I am trying to generate measurements that surpass Bad Data Detection by using an ANN with line_loading_percentages as input.

On some generated measurements i receive the error "A problem appeared while using the linear algebra methods.Check and change the measurement set." when i use the remove_bad_data function. But I do not understand why. If I only use the estimate function on the same measurement set the error is only "Estimation failed! Pandapower network failed to update!"
But if I use the chi2_analysis function I get an error stack:

```
success_chi2 = chi2_analysis(net, init="flat")
Estimation failed! Pandapower network failed to update!
Traceback (most recent call last):
  File "C:\Users\baldu\anaconda3\lib\site-packages\IPython\core\interactiveshell.py", line 3444, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-18-365dce03f2cd>", line 1, in <module>
    success_chi2 = chi2_analysis(net, init="flat")
  File "C:\Users\baldu\anaconda3\lib\site-packages\pandapower\estimation\state_estimation.py", line 151, in chi2_analysis
    return wls_se.perform_chi2_test(v_start, delta_start, calculate_voltage_angles,
  File "C:\Users\baldu\anaconda3\lib\site-packages\pandapower\estimation\state_estimation.py", line 316, in perform_chi2_test
    J = np.dot(self.solver.r.T, np.dot(self.solver.R_inv, self.solver.r))
AttributeError: 'NoneType' object has no attribute 'T'
```

Since it is probably to complex to upload my entire script, I tried to create a minimal example for you and provide the net.measurement object with the generated measurements already inserted as .json file. One file is with a generated measurement set that throws the error and I also provide a set of measurements that is also generated but that throws no error.

```
import pandapower as pp
import simbench as sb
import pandas as pd
from pandapower.estimation import estimate
from pandapower.estimation import chi2_analysis
from pandapower.estimation import remove_bad_data

grid_code = "1-LV-urban6--2-sw"
net = sb.get_simbench_net(grid_code)

measurement_df = pd.read_json("./measurements.json")
# measurement_df = pd.read_json("./measurements_working.json")
net.measurement = measurement_df

# success_chi2 = chi2_analysis(net, init="flat")
success_rn_max = remove_bad_data(net, init="flat")
# success_estimate = estimate(net, init="flat")
```
[measurements.zip](https://github.com/e2nIEE/pandapower/files/7899294/measurements.zip)

wangzhenassd, 2022-01-19:
The reason is that the state estimation algorithm failed on your measurement datasets. The convergence of the algorithm is related to multiple aspects such as initialization, measurement qualities etc.

jnsbal, 2022-01-19:
Is there any way of finding out what actually is the problem with my specified measurements? The chi2_analysis function is what I want to use. At the moment I have to catch the AttributeError in order for the script to not stop with some of the generated measurements.