e2nIEE/pandapower #1458: Inconsistent behavior of load and asymmetric_load

rbolgaryn, 2022-01-27:
I observed the following behavior of pp.runpp and pp.runpp_3ph regarding the elements load and asymmetric_load:

1) runpp

The elements load and asymmetric_load both impact the results. Having asymmetric_load elements activated increases line loading and losses. How is the algorithm considering unbalanced loads for the balanced power flow calculation?

2) runpp_3ph

Elements asymmetric_load are considered only. The elements load are completely ignored. But it would be logical to treat the load elements as if they were symmetrically configured asymmetric_load elements. 

In overall, it is inconsistent (and implicit) that in runpp the asymmetric loads are considered, but in runpp_3ph load elements are ignored.

@gourab2009 @elodin : what do you think about this?

Roman

rbolgaryn, 2022-01-27:
I noticed that load elements are considered if their "type" parameter is set to e.g. "wye".
So if the type is other than a supported value ("delta", "wye"), then the load element is ignored silently. 
I am assuming, the same is valid for sgen elements. 

Should there be an info message (if not an error) if loads have an unsupported value for type?

elodin, 2022-01-27:
Hi @rbolgaryn,
i also think it should be the same behavior for both calculations. 
How balanced pf is taking asymmetric loads into account, i don't exactly know but i suppose the three phases are being summed. I will look into it and see that both are showing consistent behavior and also implement a warning message if needed.