e2nIEE/pandapower #1475: Computing impedances using Ybus inversion: Problem with internal indices.

stephaniehoffmann, 2022-02-04:
Hi everyone, 
I am trying to compute impedances via Zbus by inverting the Ybus of a net the following way:

1. run a powerflow via runpp(net)
2. get the Ybus from ppc internal 
3. invert the Ybus
4. apply the distance resistance formula: X[i,j] = Zbus[i,i] + Zbus[j,j] - 2*Zbus[i,j].

If I have a net where the indices of the buses coincide with the internal ppc indices, this works well. 
But if the indices differ, I get strange values. 

# Example
I created a small test net, where I reproduced the problem by adding a switch, such that the internal bus indices are different. 
[image]
The black numbers are the bus indices and the red numbers are the corresponding ppc internal indices. 
For this net I would assume Line01 to have the same impedance as Line 23, but this is not the case in the resulting matrix. 
Furthermore, I wonder how to see the impedance between node 1 and 2, because there still is a connection over node 1, but the new indices identify node 1 and 2 as the same.

# Code
```
import pandapower as pp 
import pandas as pd
import numpy as np
from scipy.linalg import pinv

# create a small test net
net = pp.create_empty_network()
bus0 = pp.create_bus(net, name="Bus_0", vn_kv=110, type="n")
bus1 = pp.create_bus(net, name="Bus_1", vn_kv=110, type="n")
bus2 = pp.create_bus(net, name="Bus_2", vn_kv=110, type="n")
bus3 = pp.create_bus(net, name="Bus_3", vn_kv=110, type="n")
pp.create_ext_grid(net,bus0, vm_pu=1.02, va_degree=0)
line01 = pp.create_line(net, bus0, bus1, length_km=1.0, std_type="N2XS(FL)2Y 1x300 RM/35 64/110 kV", name="Line_0_1")
line02 = pp.create_line(net, bus0, bus2, length_km=1.0, std_type="N2XS(FL)2Y 1x300 RM/35 64/110 kV", name="Line_0_2")
line23 = pp.create_line(net, bus2, bus3, length_km=1.0, std_type="N2XS(FL)2Y 1x300 RM/35 64/110 kV", name="Line_2_3")
switch = pp.create_switch(net, bus1, bus2, et="b", type="CB", closed=True)

# run pp to create Ybus and invert Ybus
try:
    pp.runpp(net)
except pp.LoadflowNotConverged:
    pass
Ybus = net._ppc["internal"]["Ybus"]
idx_ppc = net._pd2ppc_lookups["bus"]    # to see the internal indices
Zbus = pinv(Ybus.todense())

# apply resistance distance formula
n = Zbus.shape[0]
Zbus_resistance_distance = np.zeros((n,n), dtype=np.complex128)
for i in range(n):
    for j in range(n):
        Zbus_resistance_distance[i,j] = Zbus[i,i] + Zbus[j,j] - 2 * Zbus[i,j]`
```

Do you know, how to get the proper impedances using this method, even if the internal indices differ?
Thank you!

lthurner, 2022-02-04:
I think your problem isn't about indices, it's about the electrical model of the switch.

>  For this net I would assume Line01 to have the same impedance as Line 23, but this is not the case in the resulting matrix.

The nodes 1 and 2 don't exist as seperate nodes anymore, because they are connected by a closed switch they are one and the same node electrically. So the resulting internal network has two parallel lines between the internal buses 0 and 1. The resulting admittance is defined by the parallel connection of the two original lines.

>  Furthermore, I wonder how to see the impedance between node 1 and 2, because there still is a connection over node 1, but the new indices identify node 1 and 2 as the same.

The impedances between the two nodes is by definition zero, as you wired the two nodes directly together with the switch. You can even get this electrical setup on a t-shirt: https://www.getdigital.de/widerstand-ist-zwecklos.html :)

stephaniehoffmann, 2022-02-07:
Oh I see, this makes sense. Thank you very much! :)