e2nIEE/pandapower #1486: Optimal power flow convergence issue

buddih09, 2022-02-07:
Dear all,

Could you please have a look at the test example at,
https://colab.research.google.com/drive/1BtcGjDQwxcLJGs7XY8sYJ3NHYAq7V6Ku#scrollTo=fXptnXbBELgz

Here I try to run a very simple opf problem and it does not converge. However, when I run the same problem without optimization, there is a solution. Could you please enlighten me about what might be going wrong here? Thank you very much.

buddih09, 2022-02-08:
The colab link is updated with permission to access.

SteffenMeinecke, 2022-09-12:
Sorry @buddih09, that OPF divergence issues are not really supported and unfortunately no one helped from the community.
Therefore, [I increased the importance of the warning that runopp often has divergence issues.](https://github.com/e2nIEE/pandapower/pull/1706/commits/7bc746b44216425724afb29a300576e1e044406b)

closed 2025-04-07 by vogt31337