e2nIEE/pandapower #1489: Result Sgen in Oberrhein network after powerflow is zero

Arjunayodya, 2022-02-08:
I tried some power flow calculations using Oberrhein network. after doing a power flow the result of Sgen is zero for all active and reactive power
[image]

kindly assist with why this happens?

[image]

jkisse, 2022-02-08:
This is because the `net.sgen.scaling` column is `0` by default. If you set  `net.sgen.scaling = 1` you should get the expected results in `net.res_sgen`

Arjunayodya, 2022-02-09:
Thank you for the quick reply and the solution worked

jkisse, 2022-02-09:
You're welcome. Just one more note: This is not a bug but a feature ;-) when you load the MV Oberrhein network, you can pass the argument "scenario". `net = nw.mv_oberrhein(scenario="load")` is the default and has all sgens scaled to 0. `scenario="generation"` would scale all loads to 0.1 and the sgens to 0.8 (more details [here](https://pandapower.readthedocs.io/en/latest/networks/mv_oberrhein.html))

closed 2022-02-09 by jkisse