e2nIEE/pandapower #1494: TimeSeries simulation: Just PowerFlow?

Ghoworth, 2022-02-09:
I'm trying to run a opf using time series. controller  Have all the files set up, its working on power flow.  Passed a run string ="runopf" and even "runopp" but the time series calculation is still doing an Powerflow calc and not the OPF which I know works and converges

Also When I include this line
if runstr =="runopp": ow.log_variable('res_bus','lam_p',index=[7013] ) 
its not collecting the variable lambda as it would normally do in a opf. and produces the warning 
Error at index [7013] for res_bus[lam_p]: 'lam_p'

Am I doing something wrong or is the calculation hardwired to PF.  Can i change this manually somewhere

Second question.  Can I use the output writer to capture the net.converged.  How would i do that as i dont seem to be able to get the syntax right

Thanks in advance

rbolgaryn, 2022-02-09:
```python
ow.output["Parameters"]
Out[7]: 
   time_step  controller_unstable  powerflow_failed
3          3                False             False
5          5                False             False
7          7                False             False
```

Ghoworth, 2022-02-09:
This seems to be working now - Function not string.  i could see that you use a kwargs.pop function to either use the function you pass or the default runpp function .  now it make sense.  Thanks

rbolgaryn, 2022-02-09:
You are welcome!

closed 2022-02-09 by rbolgaryn

Ghoworth, 2022-02-09:
Sorry Roman -i still cant seem to get this working it wants to run a pf not an opf

So I set 
run = pp.runopp
and then use 
timeseries.run_timeseries(net, time_steps=timerange,continue_on_divergence=True,verbose=True, kwargs=run, initial_run =True) #opf

Is it tpo do with the way im passing initial run

Ghoworth, 2022-02-09:
I guess more worrying is that im getting different results on a standalone opf vs the time series one - so it seems i have an issue with the way ive set it up.  I can see that all the correct SGENs and loads have been passed over etc.  

Gary

rbolgaryn, 2022-02-14:
No no, it is just a parameter run

..., run=pp.runopp

Ghoworth, 2022-02-16:
Thanks Roman - all good now