e2nIEE/pandapower #1496: FACTS devices model in pandapower

EngHailu, 2022-02-14:
Can anybody help me model FACTS devices paricularly UPFC in pandapower?

rbolgaryn, 2022-02-14:
We do not have an element in pandapower that would represent a UPFC device.

However, one can model such a device using the available pandapower components.

A UPFC device consists of a series part and a shunt part. For the series part, you can use an impedance element. For a shunt device, you can use a shunt element.

Then you would need to implement a controller (see tutorials for pandapower control loop). You can inherit from the pandapower Controller class and implement the methods "is_converged" and "control_step".

The implementation of the control_step should have a calculation of the required UPFC parameters/settings based on the grid state, and then adjust the parameters in net.shunt and net.impedance accordingly.

You can take a look at the transformer controllers, e.g. ContinuousTapControl, as an example of how a controller can be implemented.

Roman

vogt31337, 2025-04-02:
Closing, since it is resolved.