e2nIEE/pandapower #1504: Error in powerflow calculation at load with out of service and constant impedance current component

nitbharambe, 2022-02-19:
There seems to be a slight error in powerflow calculations under very specific conditions:

- Two loads: one in service, and one out of service are connected together.
- Any of the two loads have some constant impedance or constant current percentage component assigned.

The code to reproduce the error is as follows. The initial few lines are from the minimalist example notebook. It is compared to a copy of net with an additional `in_service=False` load: `load2` added to it. The line power outputs does not match though.

    import pandapower as pp
    
    #create empty net
    net = pp.create_empty_network()
    
    #create buses
    bus1 = pp.create_bus(net, vn_kv=20., name="Bus 1")
    bus2 = pp.create_bus(net, vn_kv=0.4, name="Bus 2")
    bus3 = pp.create_bus(net, vn_kv=0.4, name="Bus 3")
    
    #create bus elements
    pp.create_ext_grid(net, bus=bus1, vm_pu=1.02, name="Grid Connection")
    load1 = pp.create_load(net, bus=bus3, p_mw=0.100, q_mvar=0.05, name="Load", const_i_percent=0, const_z_percent=0)
    
    #create branch elements
    trafo = pp.create_transformer(net, hv_bus=bus1, lv_bus=bus2, std_type="0.4 MVA 20/0.4 kV", name="Trafo")
    line = pp.create_line(net, from_bus=bus2, to_bus=bus3, length_km=0.1, std_type="NAYY 4x50 SE", name="Line")
    
    net1 = net.deepcopy()
    load2 = pp.create_load(net1, bus=bus3, p_mw=0.100, q_mvar=0.05, in_service=False, const_i_percent=0, const_z_percent=100)
    
    pp.runpp(net, tolerance_mva=1e-8)
    pp.runpp(net1, tolerance_mva=1e-8)
    
    print("Network 1:")
    print("Trafo: p_lv_mw: %.8f, q_lv_mvar: %.8f" % (net.res_trafo["p_lv_mw"][trafo], net.res_trafo["q_lv_mvar"][trafo]))
    print("Line: p_to_mw: %.8f, q_to_mvar: %.8f" % (net.res_line["p_to_mw"][line], net.res_line["q_to_mvar"][line]))
    print("Load: p_mw: %.8f, q_mvar: %.8f" % (net.res_load["p_mw"][load1], net.res_load["q_mvar"][load1]))
    print("Network 2:")
    print("Trafo: p_lv_mw: %.8f, q_lv_mvar: %.8f" % (net1.res_trafo["p_lv_mw"][trafo], net1.res_trafo["q_lv_mvar"][trafo]))
    print("Line: p_to_mw: %.8f, q_to_mvar: %.8f" % (net1.res_line["p_to_mw"][line], net1.res_line["q_to_mvar"][line]))
    print("Load: p_mw: %.8f, q_mvar: %.8f" % (net1.res_load["p_mw"][load1], net1.res_load["q_mvar"][load1]))
    print("Load, out of service: p_mw: %.8f, q_mvar: %.8f" % (net1.res_load["p_mw"][load2], net1.res_load["q_mvar"][load2]))

Output:

    Network 1:
    Trafo: p_lv_mw: -0.10539239, q_lv_mvar: -0.05069612
    Line: p_to_mw: -0.10000000, q_to_mvar: -0.05000000
    Load: p_mw: 0.10000000, q_mvar: 0.05000000
    Network 2:
    Trafo: p_lv_mw: -0.10171551, q_lv_mvar: -0.04899518
    Line: p_to_mw: -0.09669403, q_to_mvar: -0.04834702
    Load: p_mw: 0.10000000, q_mvar: 0.05000000
    Load, out of service: p_mw: 0.00000000, q_mvar: 0.00000000

SteffenMeinecke, 2022-03-08:
Hi @nitbharambe,
thanks for reporting! I agree, it seems like an undisired behaviour and it is in pandapower at least since v2.0.0. Maybe this corner case exists from the very beginning of ZIP loads in pandapower (https://github.com/e2nIEE/pandapower/issues/20) ? - I don't know.
@jkrstulo are you available for a review?
Or is anyone else able to have a look into the code at which point the out-of-service load has an effect on the power flow calculations?

AnkurArohi, 2022-03-25:
I think this is a case of multiple mathematical solutions which lie very close to each other, can you also shouw the results of external grid in both cases?
Try to change your tolerance value maybe 1e -12 MVA

jkrstulo, 2022-04-01:
@nitbharambe, thank you for very precise code example for reproducing the bug. 
The problem was that we didn't consider `in_service` flag when calculating coefficients for constant impedance and constant current part of a voltage variable load. We considered that only for power, so this bug would not impact when only one load is at the bus. 
Sorry for my late reaction to this bug.

closed 2022-04-11 by SteffenMeinecke