e2nIEE/pandapower #1508: Double use of parameter name "type" for some elements

mstroot, 2022-02-21:
Hello dear developers,

We just stumbled into an inconsistency in either the documentation or the code itself.

In the table description of the Input Parameters for the `net.sgen` it says the folowing for `type` parameter value range:

> naming conventions:
> “PV” - photovoltaic system
> “WP” - wind power system
> “CHP” - combined heating and power system

However, in the description for the  `create_sgen` function, it says that the `type` gives the three-phase connection type ("wye" or "delta").

I am unsure wheather the first definition has any implications for any calculations or is just additional information, but using the same parameter for two different kinds of information is probably not a good idea. Especially since one must now choose between giving information about the asset type or the connection type (and therefore whether 3ph calculations are possible or not).

This might affects other elemts as well, like the asymetric sgen.

My suggestion would be to leave the asset type as the `type` parameter, since it seems to be used more widely that way and rename the 3ph connetion type for all relevant elements to something different.

Thank you for all your work on this project!

lthurner, 2022-02-21:
Hi @mstroot, I think you have analysed the situation correctly. On all elements "type" is just an "informational" parameter that doesn't impact the calculation. On asymmetric_sgen and _load, it gives the wiring type. I agree that this might be confusing. Maybe we could rename it to "connection_type"? @rbolgaryn  what do you think?

mstroot, 2022-02-22:
Thank you for the quick feedback and the clarification on the "informational" parameters. It might be helpful to identify such parameters somehow in the docs, if they aren't already. As to the renaming, as far as I am concerned that would be a good solution.

lthurner, 2022-02-22:
The parameters that are used in the documentation are marked with an asterix in the parameter table:
https://pandapower.readthedocs.io/en/develop/elements/load.html#input-parameters

But this is documented wrong here, because the type in net.load doesn't have any impact on the power flow. Only in asymmetric_load, which is a a seperate element. So the first step would be to fix the docs, and the second step to maybe rename the "type" in the asymmetric elements. Thanks for raising this issue!

rbolgaryn, 2022-02-22:
This issue is related:

https://github.com/e2nIEE/pandapower/issues/1458