e2nIEE/pandapower #1510: Internal per unit system

stephaniehoffmann, 2022-02-23:
Hi everyone, 

I am currently trying to compute impedances of a net by inverting the internal Ybus to get the Zbus. I have a question regarding the internal per unit representation:
I read in the documentation, that on a single line between two buses the unit value is the voltage of the from_bus. But how is the unit value defined between two buses, that are not directly connected?

Best regards!

vogt31337, 2025-01-03:
@stephaniehoffmann, is this still relevant? Have you found a solution?

My guess / question is what do you want to achieve? Since the per unit is for voltage only. And therefore you can lookup the voltages in the bus table. Only in the case of a line connecting two busses, a convention needs to be found, and pp uses the from_bus.

stephaniehoffmann, 2025-01-06:
Oh sorry, my problem was already solved. Kind regards.

closed 2025-01-06 by stephaniehoffmann