e2nIEE/pandapower #1518: Is there any guidance for building a three-phase network like IEEE-13

JieFeng-cse, 2022-03-03:
Hi, I want to build an IEEE-13 network for a three-phase voltage control problem, but I have no idea how to build it. I have the single-phase model, btw. I have checked the EU-LV three-phase network but it just used the json and I don't know how to build the .json from the IEEE documentation. Could you please give me some advice or is there any off-the-shelf network implementation available?
Thank you in advance!

SteffenMeinecke, 2022-03-08:
Hi @JieFeng-cse,
if you load the EU-LV three-phase network (see https://github.com/e2nIEE/pandapower/blob/develop/tutorials/ieee_european_lv_asymmetric.ipynb and https://pandapower.readthedocs.io/en/develop/networks/3phase_grids.html) you'll get the created pandapower net. A tutorial how one can create a pandapower net from scratch is presented in https://github.com/e2nIEE/pandapower/blob/develop/tutorials/create_simple.ipynb. A minimal example on unbalanced power flow is given in https://github.com/e2nIEE/pandapower/blob/develop/tutorials/unbalanced_minimal.ipynb.

JieFeng-cse, 2022-03-30:
Hi Steffen,
To be more specific, for three-phase lines we have a 3*3 impedance matrix, I don't know how to define that with pandapower. In line, it seems I could only define  r_ohm_per_km, which is a float, not a matrix. 
Thank you if you could answer my concern. @SteffenMeinecke

JieFeng-cse, 2022-03-30:
Especially for IEEE-13, it is a mixture of both single-phase and three-phase.

DanielPompermayer, 2023-01-05:
If I am not mistaken, pandapower only models three-phase transposed lines. In the line element, you can only input the positive sequence component (which is assumed to be equal to the negative one) and the zero sequence component.

If you already have the 3x3 impedance matrix (which must be the impedance matrix itself for a three-wired system or the reduced one for a three-wired system) you must check whether the values of the main diagonal are equal to each other and whether the values that are outside the main diagonal are equal to each other.

If they are so, you have a transposed line. Then you may apply the Fortescue Transform and you'll get a matrix with zeros outside the main diagonal, a value in the (0, 0) position, which is your zero component impedance, and another value in the (1, 1) and (2, 2) positions, which are you positive and negative component impedance.

If they are not so, then you do not have a transposed line. So what you could do is to approximate it to a transposed line by replacing your original matrix to that one:

| **Zm** | **Zm** | **Zm** |
| :------------: | :------------: | :------------: |
| **Zm** | **Zs** | **Zm** |
| **Zm** | **Zm** | **Zm** |

where
 
Zs = (1/3) * (Zaa + Zbb + Zcc)
Zm = (1/3) * (Zab + Zbc + Zca)

Then you will apply the Fortescue Transform and get the sequence components you will input in the line element in pandapower.

As you seem to have both single-phase and three-phase lines, I suppose you'll have to approximate your single-phase lines to three-phase three-wired lines with no interaction between the phases. I suppose you can do that by replacing the impedance matrix to:

| **Z** | **0** | **0** |
| :------------: | :------------: | :------------: |
| **0** | **Z** | **0** |
| **0** | **0** | **Z** |

Then, again, you will apply the Fortescue Transform and get the sequence components you will input in the line element in pandapower.