e2nIEE/pandapower #1522: Topological searches in pandapower

Arjunayodya, 2022-03-07:
Is there any way to get the number of radial connections from each bus point using topological searches or networkx?

SteffenMeinecke, 2022-03-08:
Hi @Arjunayodya,
yes there are topology search functionalities. Please have look to our tutorial on that: https://github.com/e2nIEE/pandapower/blob/develop/tutorials/topology.ipynb

Arjunayodya, 2022-03-08:
yes. Actually, I already gone through thus, but I coulund find anything related to get the total number of radial connections in a network, like tze bus to which the radial is connected?

rbolgaryn, 2022-03-16:
this is something you would need to implement yourself. You can look into the functions that determine cycles, connected components, etc. and find a way to use them to define what you require.

marhofmann, 2022-03-18:
@Arjunayodya you could use [shortest_path](https://networkx.org/documentation/stable/reference/algorithms/generated/networkx.algorithms.shortest_paths.generic.shortest_path.html#networkx.algorithms.shortest_paths.generic.shortest_path) starting from all nodes with ```degree == 0``` to the slack. the number of buses for all of these paths minus the number of starting nodes (```degree == 0```) should be the number you are looking for

closed 2022-05-09 by rbolgaryn