e2nIEE/pandapower #1526: optimal power flow does not converge

angelinasyrri, 2022-03-11:
Hi! I am trying to run an opf on the network attached. It does not converge! I have run diagnostics and I have changed certain impedance values, also I have rescaled the load but still it does not converge! Could you provide any idea on what could be the issue?

Simple power flow runs just fine!

[net_flex_not_converge.xlsx](https://github.com/e2nIEE/pandapower/files/8233360/net_flex_not_converge.xlsx)

buddih09, 2022-03-14:
Related to #1486

angelinasyrri, 2022-03-15:
Also, I receive the following warning when I am trying to run the opf:

C:\Users\...l\anaconda3\envs\work_env2\lib\site-packages\pandapower\pypower\pipsopf_solver.py:167: ComplexWarning:

Casting complex values to real discards the imaginary part

C:\Users\...\anaconda3\envs\work_env2\lib\site-packages\pandapower\pypower\pipsopf_solver.py:168: ComplexWarning:

Casting complex values to real discards the imaginary part

C:\Users\...\anaconda3\envs\work_env2\lib\site-packages\pandapower\pypower\pipsopf_solver.py:179: ComplexWarning:

Casting complex values to real discards the imaginary part

C:\Users\...\anaconda3\envs\work_env2\lib\site-packages\pandapower\pypower\pipsopf_solver.py:181: ComplexWarning:

Casting complex values to real discards the imaginary part

AnkurArohi, 2022-03-15:
Try to use generator which is not slack in the poly cost function, let me know if it works

angelinasyrri, 2022-03-16:
Hi! Thanks for coming back. I don't have a slack generator. I have an external grid in the topology, which is the slack bus. Do you mean  I should remove the poly cost from the external grid?

AnkurArohi, 2022-03-16:
yes remove it from external grid pls, and try with real PV nodes/generators in the network

angelinasyrri, 2022-03-16:
I have removed the ext grid from the poly cost table. But I still do not get the opf to converge. 
Are you saying that I should remove the external grid from the grid topology in general and assign another slack generator? I only have 5 PVs spread across the topology, should I add a new generator to the topology and use it as a slack generator?

AnkurArohi, 2022-03-16:
No keep the external grid , just don't use in your obj func

angelinasyrri, 2022-03-17:
Ok, I have tried this, but it does not work (No keep the external grid , just don't use in your obj func). 

I have shared the network details in my first post, this xlsx file attached on my 1st post. Can you see it?

In the meantime I have tried with PandaModels.jl and it seems to work, although I get warning 'WARNING: replacing module PP2PM', possibly because I use anaconda with spyder and I have seen in other posts that anaconda does not work well with PandaModels.jl

AnkurArohi, 2022-03-17:
There I cannot see the network connection, can you use my graph from top package of pandapower and build the network as a svg
Then I can see connections

angelinasyrri, 2022-03-18:
The network connection is in the tab 'ext_grid' of the xlsx file 'net_flex_not_converge.xlsx'. Which script are you referring to from top package of pandapower? I cannot see a file called my graph in pandapower package. With simple_plotly I can build the network attached:
[image]

JasmineGarland, 2022-05-12:
I am also having an issue with opf and the 'Casting complex values to real discards the imaginary part' warning. I also use an external grid but do not have it included in the obj func. Have you come to any conclusions as to why this is occurring?

Yiwen233, 2023-06-07:
I am facing the almost the same problem, runopp() did not converge whatever I change the net structure or load, but runpp() works well. So have you addressed this problem and how you do that?

PB-vines, 2023-09-22:
I am having the same problem. Any advice on how to fix it? @angelinasyrri @AnkurArohi @JasmineGarland @Yiwen233

keithmoffat, 2023-10-06:
Is there any resolution to this? 

I am finding that pandapower.runopp() converges for a network which has two lines in parallel between the same buses, but not for a network with a single line with twice the admittance connecting the same two buses (the two networks have the exact same admittance matrix and appropriately-adjusted power flow limits). 

Optimal power flow/OPP malfunctions should be a top priority imo.

https://pandapower.readthedocs.io/en/develop/opf/known_problems.html

vogt31337, 2025-04-02:
@angelinasyrri I will try looking into it. But the problem can be manifold, therefore OPF problems are hard to diagnose... In my experience if the normal loadflow is off by "too much" it won't work. But if you relieve the constraints it suddenly works.
So it really depends on your actual grid state. I would love to have a diagnostic function which will reveal such problems...

@keithmoffat your problem seems to be highly dependent on the actual algorithm used and the rest of the network. OPFs can be quite hard to get running on real data and not synthetic test data.

vogt31337, 2025-04-02:
Was kind a hard to get it to load, but it gave me the following output:
```
Objective Function Value = -3.47 $/hr
================================================================================
| PyPower (ppci) System Summary - these are not valid for pandapower DataFrames|
================================================================================
How many?                How much?              P (MW)            Q (MVAr)
---------------------    -------------------  -------------  -----------------
Buses           1901     Total Gen Capacity       1.3         -12.0 to 12.1
Generators      1294     On-line Capacity         1.3         -12.0 to 12.1
Committed Gens  1294     Generation (actual)     -1.1              -1.1
Loads              0     Load                     0.0               0.0
  Fixed            0       Fixed                  0.0               0.0
  Dispatchable     0       Dispatchable           0.0 of 0.0        0.0
Shunts             0     Shunt (inj)              0.0               0.0
Branches        1900     Losses (I^2 * Z)        21.12              8.36
Transformers    1900     Branch Charging (inj)     -                0.0
Inter-ties         0     Total Inter-tie Flow     0.0               0.0
Areas              1
                          Minimum                      Maximum
                 -------------------------  --------------------------------
Voltage Magnitude   0.623 p.u. @ bus 2          1.377 p.u. @ bus 1572
Voltage Angle    -372.41 deg   @ bus 1634    1105.20 deg   @ bus 1362
P Losses (I^2*R)             -                  5.26 MW    @ line 1897-1898
Q Losses (I^2*X)             -                  3.31 MVAr  @ line 1897-1898
Lambda P        -10899253.68 $/MWh @ bus 1898   9703647.05 $/MWh @ bus 606 
Lambda Q        -2910623.14 $/MWh @ bus 1470   13035647.43 $/MWh @ bus 2   

```
These values seem to be wrong, no loads and 1900 transformer? But the model doesn't include that. So there seems to be a problem in converting the model correctly. Also I had to throw away the controller, since data for them were missing...