e2nIEE/pandapower #1528: [WinError 206] The filename or extension is too long

buddih09, 2022-03-14:
I have been getting the following error when I try to import DFData. I found a [StackOverflow thread](https://stackoverflow.com/questions/682799/what-to-do-with-the-input-line-is-too-long-error-message/3583282#3583282) explaining this behavior could be due to a windows bug and provides a workaround. But I'm not exactly sure how to implement this solution with pandapower. Maybe one of you could look into it and give some  #advice? Thanks a lot. 

`Traceback (most recent call last):
  File "F:/Projects/EVA/cmab_charging_branch/cmab_charging/forecast.py", line 9, in <module>
    from pandapower.timeseries import DFData
  File "C:\Users\buddi\anaconda3\envs\cmab\lib\site-packages\pandapower\__init__.py", line 6, in <module>
    from pandapower.auxiliary import *
  File "C:\Users\buddi\anaconda3\envs\cmab\lib\site-packages\pandapower\auxiliary.py", line 43, in <module>
    from numba import jit
  File "C:\Users\buddi\anaconda3\envs\cmab\lib\site-packages\numba\__init__.py", line 19, in <module>
    from numba.core import config
  File "C:\Users\buddi\anaconda3\envs\cmab\lib\site-packages\numba\core\config.py", line 16, in <module>
    import llvmlite.binding as ll
  File "C:\Users\buddi\anaconda3\envs\cmab\lib\site-packages\llvmlite\binding\__init__.py", line 4, in <module>
    from .dylib import *
  File "C:\Users\buddi\anaconda3\envs\cmab\lib\site-packages\llvmlite\binding\dylib.py", line 3, in <module>
    from llvmlite.binding import ffi
  File "C:\Users\buddi\anaconda3\envs\cmab\lib\site-packages\llvmlite\binding\ffi.py", line 162, in <module>
    os.add_dll_directory(os.path.join(os.path.normpath(sys.prefix), 'Library', 'bin'))
  File "C:\Users\buddi\anaconda3\envs\cmab\lib\os.py", line 1105, in add_dll_directory
    cookie = nt._add_dll_directory(path)
FileNotFoundError: [WinError 206] The filename or extension is too long: 'C:\\Users\\buddi\\anaconda3\\envs\\cmab\\Library\\bin'`

SteffenMeinecke, 2022-03-15:
I guess the name of the file you are loading or writing is too long for you windows. Try another name.

SteffenMeinecke, 2023-07-12:
Ok, so the error occur if the python program needs to handle any file that path+name is too long for windows.

SteffenMeinecke, 2023-07-12:
closed due to inactivity

closed 2023-07-12 by SteffenMeinecke