e2nIEE/pandapower #1530: ext_grid vm_pu relative to what?

eddie-atkinson, 2022-03-15:
Apologies if this is a slightly silly question, but I am struggling to understand what the `vm_pu` for `create_external_grid` is in reference to?

In my example I am modelling a three-phase feeder with voltage problems and therefore the base voltage for all the buses is 433V (~1.08 in PU relative to a 230V nominal). 

Is the `vm_pu` in reference to the voltage of the bus it is connected to? 

i.e. If I had the following bus:
`b = create_bus(net, vn_kv=0.433)`
And the nominal voltage of my network was 230V would my external grid be:
`create_ext_grid(net, b, vm_pu=1.0)`
or:
`create_ext_grid(net, b, vm_pu=1.08)`

I presume my choice would influence what the `vm_pu` readings of the buses in the network would be right? 

If I used `vm_pu=1.0` a voltage of 466V on a bus would be recorded as a pu figure of 1.06. If I used `vm_pu=1.08` then a 466V on a bus would be recorded as 1.12 pu?

eddie-atkinson, 2022-03-15:
I think I figured it out myself. But wouldn't mind someone else confirming my logic.

The value you set for `vm_pu` in effect sets the baseline for the per unit system. 

For example if I attach the external grid to a bus with a voltage of 433V and set `vm_pu` to 1.04 I am in fact saying that the nominal 1.0 pu voltage is 416V.

Whereas if I attach the external grid to a bus with a voltage of 433V and set `vm_pu` to `1.0` I am saying that the nominal voltage is 433V

AnkurArohi, 2022-03-15:
Hey I can just share what I understand, so basically vm_pu refers to the voltage magnitude defined at the slack bus with reference to the base voltage in pu.

So if your base voltage is 10kV, then you define the magnitude of voltage at this slack bus to be x pu
If x=1.04
Then you have the voltage as 10.4kV
And because it is a slack bus it will be kept constant after calculation

eddie-atkinson, 2022-03-16:
@AnkurArohi 

If I had a network which had buses with a nominal voltage of 416V, but for my simulation wanted the buses to be at 433V would I set `vn_kv=0.416` for the buses and `vm_pu=1.04` to give a bus voltage of 433V (assuming no loads, generators or line losses).

AnkurArohi, 2022-03-16:
You need to compare that with your base voltage, so if your base is 416 then keep buses at 433/416

AnkurArohi, 2022-03-16:
0.416 is wrong

eddie-atkinson, 2022-03-17:
> You need to compare that with your base voltage, so if your base is 416 

Where does your base voltage come from? From reading the documentation I feel like it is the bus connected to the external grid.

I think I haven't given enough context to adequately describe what I am trying to do.

I am attempting to train reinforcement learning agents to moderate voltage between 0.9 and 1.10 pu. In order to train them I am modelling a real network which runs at 1.04 pu, relative to a baseline of 416V.

Therefore, I want the buses of this network to be at 1.04 pu before we add loads and generators. To achieve I thought I needed to set the reference voltage of all the buses to 0.416 kV and then supply power at 1.04 pu from the external grid? 

In my testing this seems like it produces the right result?
[image]

rbolgaryn, 2022-03-25:
You need to set buses vn_kv = 400 (or 380, 110, etc.), and then supply power at ext_grid with 1.04.

The voltage at the ext_grid bus will be 400 * 1.04, and the voltage at other buses will vary based on the power flow.

eddie-atkinson, 2022-03-27:
Thanks @rbolgaryn that makes sense :)