e2nIEE/pandapower #1535: infofunc in collections (line, bus, switches)

Arjunayodya, 2022-03-21:
Hi all.

I tried to include the name of bus, line and switch in the plot using collections. There is an input 'infofunc', even though I created a function with bus and line information, the code is working but no information is displayed on the plot

Could someone explain what is the 'Infofunc' means?

Is there any way to annotate/hover the bus, line and switch information (like name or index) on the plot using collections  ??

jkisse, 2022-06-10:
Hi,
the infofunc is only suitable for plotly-plots and provides the hoverinfo for the different elements. If you want to add labels to a "simple_plot" or any other matplotlib / collection based plot, have a look at the [plotting tutorial](https://github.com/e2nIEE/pandapower/blob/develop/tutorials/plotting_basic.ipynb). It provides examples for annotated and highlighted elements.

SteffenMeinecke, 2023-07-12:
closed due to inactivity

closed 2023-07-12 by SteffenMeinecke