e2nIEE/pandapower #1549: Distributed slack feature results for generators 3are misguiding

AnkurArohi, 2022-03-25:
After the distributed slack feature is toggled on and provided with slack weights, the result tables of the generators display "va_degree " value also zero which is actually not the case as in the Jacobian matrix implementation of distributed slack feature there is an actual slack(ref_slack) with va_degree zero used and the other generators are only used to rescale/redistribute this base_slack power / power mismatch to all generators including the base slack depending on the slack weight parameter.

This means they are a part of Active power redistribution but they are not a slack node in itself.

Reference for this is the following closed ticket =https://github.com/e2nIEE/pandapower/issues/1408

Hence as a suggestion I would request a new column in the result table of the generators:-
net.res_gen["ref_Slack"] which should be True just in the case of base_slack whose power has been redistributed
Then only in this case it should be allowed to have a "va_degree" = 0

This will make the result table non misleading.

rbolgaryn, 2022-03-25:
Hi @AnkurArohi ,

can you please provide a minimal example to illustrate the issue?

Roman

AnkurArohi, 2022-03-29:
Yes sure

AnkurArohi, 2022-03-30:
So, here is what i have done , First I have loaded the network and then run following
This pandapower network includes the following parameter tables:
   - bus (2935 elements)
   - load (67 elements)
   - sgen (499 elements)
   - gen (6 elements)
   - switch (3199 elements)
   - ext_grid (1 element)
   - line (47 elements)
   - trafo (49 elements)
   - trafo3w (13 elements)
   - impedance (38 elements)
   - trafoend (137 elements)
   - busbarsection (173 elements)

In [22]: net.gen.slack = True

In [23]: net.gen.slack_weight = 1.0

In [24]: net.gen
Out[24]: 
                                        uuid ems__entity_id   bus  p_mw  q_mvar     vm_pu  ...  scaling                      type  slack  slack_weight  controllable  topo_island
index                                                                                      ...                                                                                   
0      _0815b4c3-bfae-59a5-b169-760ba848375d           None   273   0.0     0.0  1.018624  ...        1  ExternalNetworkInjection   True           1.0          true            0
1      _264ff49f-b61d-5558-8045-ec6c3bf9a906           None   563   0.0     0.0  1.052365  ...        1  ExternalNetworkInjection   True           1.0          true            0
2      _a4cd4e4b-43d4-5a0a-ab68-8cd87aa35b84           None   848   0.0     0.0  1.034897  ...        1  ExternalNetworkInjection   True           1.0          true            0
3      _86d03500-ffa2-5fdc-815c-426fddd53106           None  1197   0.0     0.0  1.052365  ...        1  ExternalNetworkInjection   True           1.0          true            0
4      _f9299609-e75c-51c1-bc0c-32b65834347c           None  2032   0.0     0.0  1.018650  ...        1  ExternalNetworkInjection   True           1.0          true            0
5      _a34ff8c0-31be-54d1-b7a9-4cc709df6ebf           None  2458   0.0     0.0  1.034897  ...        1  ExternalNetworkInjection   True           1.0          true            0

[6 rows x 20 columns]

In [25]: try_running_pp(network, algorithm=pf_algorithm, log=log)
Now analysed the results:-

            p_mw    q_mvar  va_degree     vm_pu
index                                          
0     -10.588371  0.661153        0.0  1.018624
1       6.730835 -1.051909        0.0  1.052365
2      -0.087772 -0.285877        0.0  1.034897
4      -2.116224  0.136628        0.0  1.018650
5      -0.087772 -0.285877        0.0  1.034897

As you see the angle of all the generators are zero whicch is obviously not correct as in the Jacobian matrix there is a base_slack which has this voltage angle 0 (only)

AnkurArohi, 2022-04-01:
@rbolgaryn

AnkurArohi, 2022-04-06:
Hence the issue is that there is no clear demarcation between active power balancing PV bus generators a d a real slack@rbolgaryn

rbolgaryn, 2022-04-13:
sorry I am having a hard time with understanding this example. Can you please provide a small, reproducible test example?

AnkurArohi, 2022-04-25:
@rbolgaryn 
Since the network is huge, it is very diffucult for me to provide a smaller, easily understandable code

rbolgaryn, 2022-06-09:
have you taken a look at the jupyter notebooks?

Can you point out the inconsistency using an example from the jupyter notebook?

https://github.com/e2nIEE/pandapower/blob/develop/tutorials/distributed_slack.ipynb

AnkurArohi, 2022-06-16:
> have you taken a look at the jupyter notebooks?
> 
> Can you point out the inconsistency using an example from the jupyter notebook?
> 
> https://github.com/e2nIEE/pandapower/blob/develop/tutorials/distributed_slack.ipynb

Yes I am on it, thanks

closed 2024-11-22 by pawellytaev