e2nIEE/pandapower #1551: Unbalanced 3ph timeseries powerflow errors

eddie-atkinson, 2022-03-29:
Hi, 

I am having issues running a 3 phase timeseries powerflow. I keep getting errors related to elements that appear in balanced powerflows such as:
```
Error at index [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29] for res_line[loading_percent]: "None of [Int64Index([ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15, 16,\n            17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29],\n           dtype='int64')] are in the [index]"
```
and
```
Error at index [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30] for res_bus[vm_pu]: "None of [Int64Index([ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15, 16,\n            17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30],\n           dtype='int64')] are in the [index]"
```

My understanding is that these elements are not even present in an unbalanced powerflow. When I run the powerflow statically these elements are not present in the results tables.

I have found that I can still run the timeseries powerflow in spite of these errors and I get results at the end. I would guess that these errors don't affect the output of the timeseries simulation?

Here is an MRE:
```
import pandapower as pp
import pandapower.control as control
import pandapower.timeseries as timeseries
from pandapower.timeseries.data_sources.frame_data import DFData
from pathlib import Path
import pandas as pd

net = pp.from_pickle(Path("./model.p"))
solar_df = pd.read_csv(Path("./test_pv.csv"))
# We don't really care about datetime for this example
solar_df = solar_df.drop(columns=["datetime"])
ds_solar = DFData(solar_df)

# We want to find the sgens connected to the A phase to assign their generation to
# Bit of a hack, but using the name of the generator is an easy way to assign that
sgen_mw_a_index = net.sgen[net.sgen["name"] == "a"].index

const_sgen_a = control.ConstControl(
    net,
    element="asymmetric_sgen",
    element_index=sgen_mw_a_index,
    variable="p_a_mw",
    data_source=ds_solar,
    profile_name=[str(i) for i in sgen_mw_a_index],
)

ow = timeseries.OutputWriter(
    net, output_path=Path("./sim_results"), output_file_type=".csv", csv_separator=","
)

ow.log_variable("res_asymmetric_sgen_3ph", "p_a_mw")
ow.log_variable("res_bus_3ph", "vm_a_pu")

# Don't think I need this line, but keep it just in case
pp.toolbox.create_continuous_elements_index(net)

timeseries.run_timeseries(net, run=pp.runpp_3ph)

```

Please see the attached zip for the model and `test_pv.csv` file.
[Archive.zip](https://github.com/e2nIEE/pandapower/files/8371906/Archive.zip)

If someone can pinpoint the issue I'm happy to contribute a PR fixing the problem. Lord knows I've had enough free support from you guys and should contribute something back 😇

jurasofish, 2022-04-09:
Looks like OutputWriter defaults to logging res_bus.vm_pu and res_line.loading_percent (https://pandapower.readthedocs.io/en/v2.9.0/timeseries/output_writer.html#output-writer)

You could use `OutputWriter.remove_log_variable` to remove them, or pass `log_variables` when initialising OutputWriter.

the particular error occurs here https://github.com/e2nIEE/pandapower/blob/28c2c7eb87e7c6f709e0c344f02b0c0619d99742/pandapower/timeseries/output_writer.py#L470

ps some would consider it rude to ask strangers to handle your pickle 😳

eddie-atkinson, 2022-04-09:
Thanks, @jurasofish. 

Apologies for the pickle, should have just made a simpler network

jurasofish, 2022-04-09:
No worries. In case you're not aware - what I was getting at is that unpickling can execute arbitrary code so is a security risk