e2nIEE/pandapower #1553: inconsistent element columns behaviour

SteffenMeinecke, 2022-03-31:
There are some inconsistencies with element columns of new features.
My proposal is that columns should be either (1) mandatory (for power flow) or (2) optional.
If (1), columns should be available in `pp.create_empty_network()`, after `convert_format()` and thus also in the nets received from `pandapower.networks`.
If (2), columns shouldn't be in `pp.create_empty_network()` and shouldn't be added by any `create` function if no such parameters are passed.

However, as far as I observed,

1. `net.trafo.pt_percent` and `net.trafo.oltc` are created within the create function without passed data
2. `net.gen.power_station_trafo` is availble after `pp.create_empty_network()` but not after `convert_format()`.

Maybe there are some more inconsistencies.
Generally, a helper function that ensures necessary columns and correct dtypes depending on the type of calculation (power flow, optimal power flow, short-circuit, ...) would be great.

pawellytaev, 2024-11-22:
we should discuss this and add this to the list in #1628

heckstrahler, 2026-11-10:
with introduction of pandera in pandapower 4.0.0 this has been completed

closed 2026-11-10 by heckstrahler