e2nIEE/pandapower #1555: Powerflow direction

yasirroni, 2022-03-31:
It is nice to see [load loading in color map](https://github.com/e2nIEE/pandapower/blob/dde080ba1a86bb042085852c9bcd3546e0fee6f5/tutorials/plotting_colormaps.ipynb)

Based on [basic plotting](https://github.com/e2nIEE/pandapower/blob/dde080ba1a86bb042085852c9bcd3546e0fee6f5/tutorials/plotting_basic.ipynb), we can add bus indexing.

Can you show an indexing showing the direction of power flow and their value? Thank you.

rbolgaryn, 2022-04-10:
an easy way to do it would be to plot voltage angle (this approach is useful in transmission grids)

yasirroni, 2022-04-10:
> an easy way to do it would be to plot voltage angle (this approach is useful in transmission grids)

Have you implemented it?

rbolgaryn, 2022-04-11:
no

closed 2024-11-22 by pawellytaev