e2nIEE/pandapower #1556: [MATPOWER-PIP Maintainer]

yasirroni, 2022-03-31:
Hello, pandapower maintainer. I'm from matpower-pip.

From the docs, pandapower only support conversion from .mat file. Here, I want to share an example how to do a conversion from .m file via oct2py. Sadly, it need a third parties octave.cli. See [here](https://github.com/yasirroni/pandapower-short-circuit-analysis/blob/main/notebooks/1.power_flow.ipynb)

Maybe someone read this issue can get some benefit,

I'm getting a trouble how to add some more data to run short circuit analysis on matpower case thought.
_
Another approach is using [psst](https://github.com/ames-market/psst) matpower case parser. I'm on my way to refactor that and extract only the converter and display it as pandas dataframes.

SteffenMeinecke, 2022-04-11:
Hi @yasirroni,
many thanks for raising that links. Your option to load .m files via `matpower-pip` and `oct2py` is great. Nevertheless, I guess that most users how are not happy with the current solution since they are not able to provide a `.mat` file from their `.m` file would like to have a simpler solution than first installing a non-python piece of software like octave. But it is great to know that this option exist!
The PSST parser is more what I prefer as a solution. Since you are already reviewing that:
Is their license open to reuse their parser in pandapower? 
Can you post a minimal example code how to use their parser or post which functions of which files are promising to be reused?

yasirroni, 2022-04-11:
```
from matpowercaseframes.core import CaseFrames

case_path = 'case9.m'
cf = CaseFrames(case_path)

print(cf.gencost)
```

yasirroni, 2022-05-23:
Any news?

I will be very glad if you just use my packages instead of rewriting. Or you need any help? Should I make a PR?

___

Furthermore, using `matpower-pip`, usage will be as easy as:
```
import os
from matpower import path_matpower # require `pip install matpower`
from matpowercaseframes import CaseFrames

case_name = 'case9.m'
case_path = os.path.join(path_matpower, 'data', case_name)
cf = CaseFrames(case_path)

print(cf.gencost)
```

SteffenMeinecke, 2022-06-02:
@yasirroni you may have a look at https://github.com/e2nIEE/pandapower/pull/1598. If you aggree on that proposed implementation, you can close this issue and I merge the PR. If not, please, let me know.

yasirroni, 2022-06-02:
That is cool. But, I'm not reviewing in details on how dataframe from matpowercaseframes converted to pandapower format.

I hope this will motivate me more to improve the performance of matpowercaseframes. Thank you!

closed 2022-06-02 by yasirroni