e2nIEE/pandapower #1558: get_connected_elements always returns empty set although there are elements connected

AnkurArohi, 2022-04-01:
pp.get_connected_elements(merged_network,"l",116648,respect_switches=False)
Out[46]: set()
merged_network.bus.loc[merged_network.bus.uuid=="_BAD1A7B7-01AB-58AE-E054-0010E06FFF68"]
Out[47]: 
                                         uuid  ... foreign_net
1303    _BAD1A7B7-01AB-58AE-E054-0010E06FFF68  ...       False
116648  _BAD1A7B7-01AB-58AE-E054-0010E06FFF68  ...        True
[2 rows x 23 columns]
pp.get_connected_elements(merged_network,"l",1303,respect_switches=False)
Out[48]: set()

rbolgaryn, 2022-04-10:
Hi @AnkurArohi ,

please provide a reproducible test function.

rsparks3, 2024-01-12:
Hello! I am having the same issue.  Here is code that reproduces the issue with a pandapower example network.  This code takes the first bus of the example_simple network and should return elements connected to it, but it returns an empty list.  

```
net = pp.networks.example_simple()
bus = list(net.bus.iloc[0]['name'])
print(pp.get_connected_elements_dict(net,bus))
```

closed 2024-11-22 by pawellytaev