e2nIEE/pandapower #1568: Asymmetric 3ph timeseries resolving results in KeyError

JimiC, 2022-04-13:
Hi, again.

We are trying to resolve asymmetric timeseries for a network using simply the `run_timeseries` function.

We have stubbled on, what we believe is a minor bug but a bug non the less.
Running a timeseries that does not converge (we know it doesn't) with option "continue_on_divergence=False", a "LoadflowNotConverged" exception is raise, which is expected and handled from our side. But if we use "continue_on_divergence=True" then the tool crashes with a `KeyError` exception in [_evaluate_net](https://github.com/e2nIEE/pandapower/blob/v2.9.0/pandapower/control/run_control.py#L197) simply because the `net['converged']` is `False` and when trying to evaluate the `net['OPF_converged']` it doesn't find the `OPF_converged` key.

Bug is fixed if you simply add a guard clause for the `OPF_converged` key like:
`ctrl_variables['converged'] = net['converged'] or ('OPF_converged' in net and net['OPF_converged']`)`

Thanks in advance,
Jimi from ICCS

JimiC, 2022-04-13:
After some more code digging I believe the best way to resolve this is to add `"OPF_converged": False` in `create.py` at https://github.com/e2nIEE/pandapower/blob/v2.9.0/pandapower/create.py#L506

rbolgaryn, 2022-05-16:
I agree that OPF_converged in create is a better option

rbolgaryn, 2022-05-16:
Dear @JimiC ,

can you please write a small test for the case you described?

Roman

JimiC, 2022-05-16:
Not sure if it's necessary as the OP error was a `KeyError`, simply because the `OPF_converged` key was not in the dictionary. Adding it resolves all issues that may occur with accessing the `OPF_converged` value.
From what I saw in the code `OPF_converged` is not initialized if you are running `runpp_3ph`.

But if you insist, I can try. Is `pandapower/test/timeseries/test_timeseries.py` the correct file to add the test cases?

rbolgaryn, 2022-05-17:
it is added in line 56 of powerflow.py

rbolgaryn, 2022-05-17:
@elodin can you please add the initialization in 3ph powerflow?

rbolgaryn, 2022-05-17:
in the same pull request

JimiC, 2022-05-17:
> it is added in line 56 of powerflow.py

`_powerflow` in `powerflow.py` is not called if you run `runpp_3ph`. Instead `runpp_3ph` is called from `runpp_3ph.py`.
Also when the code-flow goes through `net_initialization` in `run_control.py`, if `initial_run` is `False`, `net["converged"] = True` is set on line 164 which bypass the issue. If instead `initial_run` is `True` then the `KeyError` would appear there.
In our case the code flow blows up when it reaches `_evaluate_net` at line 197 where the same condition appears and if `net['converged']` is `False`.

JimiC, 2022-05-17:
From my POV, if you add `OPF_converged` in `create.py` you can avoid blotting the code-base with initializing it in several places and it's set to `True` whenever it makes sense.

elodin, 2022-06-08:
So if i understand you correctly @JimiC, adding the initialization in `create.py` did already fix the problem?
During my testing, `OPF_converged` was initialized every time, because `create_empty_network` is always executed no matter how the net is built. If this is correct, i'd rather not add the initialization in runpp_3ph another time @rbolgaryn and would suggest to remove it from `powerflow.py` too, to avoid clutter.

JimiC, 2022-06-08:
@elodin Issue has been fixed since 5c4c47561c059a8f7cf917f19959e3cd8e9ebd99 but I'm waiting for the release to close this issue.

JimiC, 2022-08-30:
This fix has been released. Thanks.

closed 2022-08-30 by JimiC