e2nIEE/pandapower #1571: Casting of results after DC power flow leads to Type error

AnkurArohi, 2022-04-19:
pandapower/results_bus.py", line 438, in _get_shunt_results
    bus_pq[b_ppc, 0] += vp
TypeError: ufunc 'add' output (typecode 'O') could not be coerced to provided output parameter (typecode 'd') according to the casting rule ''same_kind''

rbolgaryn, 2022-04-19:
Dear @AnkurArohi ,

please provide a reproducible minimal example to test this behavior.

Best regards,
Roman

AnkurArohi, 2022-04-25:
@rbolgaryn 
My problem is that the network is really big and cumbersome to get to this point

AnkurArohi, 2022-04-25:
But I have found a solution to this,
In the python version 2.7
We have this error in the newer versions actually not anymore
The solution for the older version was to use this
bus_pq[b_ppc, 0] =bus_pq[b_ppc, 0] +vp
instead of the previous form

rbolgaryn, 2022-05-02:
Oh, but the Python 2.x versions are no longer supported. Please use the supported version of Python (you can see which Python versions are supported in the GitHub page or PyPI page)

closed 2022-05-02 by rbolgaryn