e2nIEE/pandapower #1573: Y bus calculation from ppc

Arjunayodya, 2022-04-25:
I am going through the code of PPC, y bus creation an d so on.

I have considered the example 'panda power.networks.example_simple() for this with 7 buses, 4 lines and 1 transformer.

For Y bus, I am getting nb (no of bus)=net._ppc['bus']= 8 (but total no of bus is 7 in-network)

Why is it so?

AnkurArohi, 2022-04-25:
Hi @Arjunayodya  Please provide a reproducible code.
So that we could test it and see for ourselves 
thanks

Arjunayodya, 2022-04-25:
import pandas as pd
import pandapower as pp
net=pandapower.networks.example_simple()
pp.runpp(net)

bus=net._ppc['bus']

nb=bus.shape[0]

ascheidl, 2022-04-25:
have a look at 

https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb

closed 2022-06-15 by rbolgaryn