e2nIEE/pandapower #1576: Not a Bug:- Is it possible to perform Network Reduction in pandapower

AnkurArohi, 2022-04-28:
I would like to get some idea about network reduction , is there any module or any test or code snippet where in network reduction has been performed on the basis of voltage level  so to include all elements in the network wchich only belong to one voltage level?
Basically to replace the trafo with a PV gen or PQ sgen node?

Thanks well in advance

rbolgaryn, 2022-05-03:
we just released this functionality to the develop branch, in pandapower.grid_equivalents

It would be very helpful if you try applying it and let us know your feedback.

Best regards, Roman

AnkurArohi, 2022-05-04:
@rbolgaryn  Thank you 
What kind of network reductions are supported I mean at what level of granularity?
Are we substituting transformers with network element if yes which one and with what parameters?

It would be nice if you could share PR for this feature

rbolgaryn, 2022-05-04:
Have a look at the tutorial:

https://github.com/e2nIEE/pandapower/blob/develop/tutorials/grid_equivalents.ipynb

AnkurArohi, 2022-06-16:
Yes there are issues in understanding what I mean by network reduction and what is the aim in the commit

AnkurArohi, 2022-06-16:
To understand this better I would request a meeting? what do you say? like online in zoom or any other app, max 30 mins would be enough

AnkurArohi, 2022-06-16:
But @rbolgaryn  I had a quick look in the grid equivalent script!
Good work!!!

SteffenMeinecke, 2023-07-12:
closed due to inactivity

closed 2023-07-12 by SteffenMeinecke