e2nIEE/pandapower #1589: Disable automatic browser opening | plotly

jwiemer112, 2022-05-19:
The draw_traces opens the plotly plot in a browser and returns the fig,

```
    if _in_ipynb():
        from plotly.offline import init_notebook_mode, iplot as plot
        init_notebook_mode()
        plot(fig, filename=filename)
    else:
        from plotly.offline import plot as plot
        plot(fig, filename=filename, auto_open=auto_open)

    return fig
```

I think it's better to only return the fig, so the user can decide if he want's to show the figure or just to save it.

You can just call `pf_res_plotly(net, on_map=False).show(renderer="browser")` for example to open the fig directly in the browser.

What do you think about it? 

Greetz Janni :)

jkisse, 2022-05-19:
Could you please specify how this would differ from the already implemented `auto_open=False` which would just save the figure and not open it in the browser?

vogt31337, 2022-05-19:
auto_open -- If True, open the saved file in a web browser after saving.

Would be okay, but you can't suppress the file saving. So in every case it writes a html file: auto_open=True, a browser get's opened. =False, file get's written no browser is opened. Is quite annoying that it always writes a file...

Maybe I'm using it wrong.

jwiemer112, 2022-05-19:
We could just return the figure and let the user decide what to do with it, maybe that's the best solution?

jkisse, 2022-05-20:
For continuity in the API, I would not change the default simple_plotly behavior, but prefer another keyword argument (maybe just in **kwargs). For example, the plotly `plot` function (in which the file is created) has an[ `output_type` parameter ](https://github.com/plotly/plotly.py/blob/master/packages/python/plotly/plotly/offline/offline.py#L449) that is `file` by default. If it is set to `div`, no file will be created. This parameter could be passed from the simple_plotly call all the way through to the [plot() call](https://github.com/e2nIEE/pandapower/blob/07eeae90e0f6e3f4920089dca97d92f9a1de8ea3/pandapower/plotting/plotly/traces.py#L912). PRs welcome ;-)

l1358, 2022-10-17:
Hi @jwiemer112 
I have a function which returns a pf_res_plotly figure and I need to show that figure on a PYSimpleGui canvas. 
When I call pf_res_plotly(net, on_map=False).show(renderer="browser") it works perfectly but I have no success showing the figure on a canvas. Do you have any suggestion?
Thanks

jwiemer112, 2022-11-01:
Hey, 

maybe this helps:

https://plotly.com/python/renderers/

jkisse, 2024-12-04:
Hi @jwiemer112 , in order to close this issue soon, I'd propose to implement your suggested `pf_res_plotly(net, on_map=False).show(renderer="browser")` and simply update the docstring, pointing out the change and what the users need to know now to apply the new approach. Could you prepare a PR?

jkisse, 2024-12-19:
For now, the solution on my fork is [this](https://github.com/e2nIEE/pandapower/compare/develop...jkisse:pandapower:transhyde#diff-9907702c93497dec0274959e13e5c9b92b27b92bcbf29f453387f7df0942c9b5R1212-R1216): if `filename` is `None`, no file is created and just the figure is returned. 

```
def draw_traces(...)
        ...
        if _in_ipynb() and auto_open:
            from plotly.offline import init_notebook_mode, iplot as plot
            init_notebook_mode()
            plot(fig, filename=filename)
        elif filename is not None:
            from plotly.offline import plot as plot
            plot(fig, filename=filename, auto_open=auto_open)
      # else: just return the fig
    return fig
```
Shall we adapt that in the main pandapower library, @jwiemer112 @vogt31337 ?