e2nIEE/pandapower #1590: Problem with defining angles when using ext_grid

shamigulov, 2022-05-19:
_#I have this simple code:_

import pandapower as pp
net = pp.create_empty_network() 
b1 = pp.create_bus(net, vn_kv=10.)
b2 = pp.create_bus(net, vn_kv=10.)
pp.create_line(net, from_bus=b1, to_bus=b2, length_km=30, std_type="NAYY 4x50 SE")
pp.create_ext_grid(net, bus=b1, **va_degree= -1.0**, slack_weight= 1)
pp.create_ext_grid(net, bus=b2, **va_degree= 12.0**, vm_pu= 1.0, slack_weight= 1)
pp.runpp(net, calculate_voltage_angles= True, distributed_slack= True)
print('buses:\n',net.res_bus)
print('_'*150, '\nlines:')
print(net.res_line)
print('_'*150, '\next_grids:')
print(net.res_ext_grid)

_#And I get this result:_

buses:
    vm_pu  va_degree          p_mw   q_mvar
0    1.0       **-1.0** -3.478121e-16  0.09896
1    1.0       **-1.0**  3.812055e-16  0.09896
______________________________________________________________________________________________________________________________________________________ 
lines:
      p_from_mw  q_from_mvar       p_to_mw  q_to_mvar         pl_mw  ql_mvar  \
0  3.478121e-16     -0.09896 -3.812055e-16   -0.09896 -3.339343e-17 -0.19792   

   i_from_ka   i_to_ka      i_ka  vm_from_pu  va_from_degree  vm_to_pu  \
0   0.005713  0.005713  0.005713         1.0            -1.0       1.0   

   va_to_degree  loading_percent  
0          -1.0         4.023569  
______________________________________________________________________________________________________________________________________________________ 
ext_grids:
           p_mw   q_mvar
0  3.478121e-16 -0.09896
1 -3.812055e-16 -0.09896

_"""Why does the second bus get the wrong angle - (**-1.0**), like on the first bus? This should result in **12.0**.
And, accordingly, the calculation of the power flow from the first bus to the second is completely wrong. There is practically no power flow, although it should be quite significant.
The link to Colab  [https://colab.research.google.com/drive/1Vrj4ycpZ_TIfyHWsW8CVcmi8T0UD_WQi?usp=sharing](url)  """_

SteffenMeinecke, 2022-06-02:
Dear @shamigulov,
the power flow between two ext_grids is already fixed by its angles. One cannot consider both, the fixed angles *and* the constraint that the slack power should be distributed as defined by the factor `slack_weight`.
If you want the first, run the powerflow without distributed slack. If you want the second, using just one ext_grid and other generation elements as gen with values for `slack_weight` is a good idea. Furthermore, investigating a grid without loads and thus power flows in value range of 1e-16 can possibly raise issues which you do not want to investigate.

shamigulov, 2022-06-21:
**Ok. I'm tried like this:**

import pandapower as pp
net = pp.create_empty_network() 
b1 = pp.create_bus(net, vn_kv=10.)
b2 = pp.create_bus(net, vn_kv=10.)
pp.create_line(net, from_bus=b1, to_bus=b2, length_km=30, std_type="NAYY 4x50 SE") 
pp.create_ext_grid(net, bus=b1, va_degree= -1.0, slack_weight= 1)
pp.create_ext_grid(net, bus=b2, va_degree= 12.0, vm_pu= 1.0, slack_weight= 1)
pp.runpp(net, calculate_voltage_angles= True, distributed_slack= False)
print('buses:\n',net.res_bus)
print('_'*150, '\nlines:')
print(net.res_line)
print('_'*150, '\next_grids:')
print(net.res_ext_grid)

**Result:**

buses:
    vm_pu  va_degree  p_mw   q_mvar
0    1.0        0.0   0.0  0.09896
1    1.0        0.0   0.0  0.09896
______________________________________________________________________________________________________________________________________________________ 
lines:
   p_from_mw  q_from_mvar  p_to_mw  q_to_mvar  pl_mw  ql_mvar  i_from_ka  \
0        0.0     -0.09896      0.0   -0.09896    0.0 -0.19792   0.005713   

    i_to_ka      i_ka  vm_from_pu  va_from_degree  vm_to_pu  va_to_degree  \
0  0.005713  0.005713         1.0             0.0       1.0           0.0   

   loading_percent  
0         4.023569  
______________________________________________________________________________________________________________________________________________________ 
ext_grids:
   p_mw   q_mvar
0   0.0 -0.09896
1   0.0 -0.09896

**Both angles are becoming zeros. Why? **

AnkurArohi, 2022-06-21:
There is no power flow. Please define the network correctly, you have two buses and both have external grids. Please define another bus with a load or so

shamigulov, 2022-06-21:
Thanks for your reply. Of course, I will try to do as you write.
But can't power flow from one external network to another? Without any other elements of the power system in the area under consideration.

shamigulov, 2022-06-22:
> If you really want to simulate two external grids connected to each other then you need to make distributed_slack = True

Please see my first example (at the top of the issue). If I understand correctly, that's exactly what I did. But the simulation result was wrong.

rbolgaryn, 2022-06-30:
you can contribute a failing tets in a pull request

closed 2026-11-10 by heckstrahler