e2nIEE/pandapower #1591: Integrating power-grid-model into pandapower

TonyXiang8787, 2022-05-19:
**Integrating power-grid-model into pandapower**

Mainly to profit the performance benefit of power-grid-model.

https://github.com/alliander-opensource/power-grid-model
https://github.com/alliander-opensource/power-grid-model-benchmark/blob/main/Power%20Grid%20Model%20Benchmark.ipynb

**Integrating method**

- A conversion script which converts `pandapower` input to `power-grid-model` input, and converts `power-grid-model` output to `pandapower` output.
- Add an additional argument `use_power_grid_model` and `threading` in the function `runpp` and `runpp_3ph`.
- Add an additional argument `use_power_grid_model` and `threading` in the function `run_timeseries`.
- Add a function which does N-1 scenario calculation in `pandapower` which calls `power-grid-model`.

SteffenMeinecke, 2023-07-12:
the integration to the documentation is still missing

TonyXiang8787, 2023-07-20:
Documentation has been added. This issue can be closed.